sidoh/epaper_templates #1: Compile error

StofLE, 2018-09-07:
Hi, nice project!

I'm on Windows with Visual Studio Code and PlatformIO and get compile error:

```
Processing esp32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
INFORMATION: Es konnten keine Dateien mit dem angegebenen
Muster gefunden werden.
PLATFORM: Espressif 32 > DOIT ESP32 DEVKIT V1
SYSTEM: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny)
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(deep) COMPATIBILITY(soft)
Collected 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit GFX Library> 1.2.9 #5b672ff
|   |-- <SPI> 1.0
|-- <GxEPD> #5c075bc
|   |-- <Adafruit GFX Library> 1.2.9 #5b672ff
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|-- <Timezone> #bbea256
|   |-- <Time> 1.5
|-- <WifiManager> 0.99.9 #10406e9
|   |-- <DNSServer> 1.1.0 #9113193
|   |   |-- <WiFi> 1.0
|   |-- <WebServer> 1.0 #75d2645
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0 #9113193
|   |-- <WiFi> 1.0
|-- <WebServer> 1.0 #75d2645
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <AsyncMqttClient> 0.8.2 #33cdd50
|   |-- <AsyncTCP> 1.0.1
|-- <ESP Async WebServer> 1.2.0
|   |-- <AsyncTCP> 1.0.1
|   |-- <Time> 1.5
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 5.13.2
|-- <ArduinoJson> 5.13.2
|-- <Time>
|   |-- <Timezone> #bbea256
|   |   |-- <Time> 1.5
|-- <NTPClient> 3.1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <HTTP>
|   |-- <TokenParsing>
|   |-- <FS> 1.0
|   |-- <ArduinoJson> 5.13.2
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <Update> 1.0
|   |-- <ESP Async WebServer> 1.2.0
|   |   |-- <AsyncTCP> 1.0.1
|   |   |-- <Time> 1.5
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ArduinoJson> 5.13.2
|   |-- <Display>
|   |   |-- <DataStructures>
[... 219 lines cut ...]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_STA_LOST_IP' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_STA_WPS_ER_SUCCESS' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_STA_WPS_ER_FAILED' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_STA_WPS_ER_TIMEOUT' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_STA_WPS_ER_PIN' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_AP_START' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_AP_STOP' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_AP_STACONNECTED' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_AP_STADISCONNECTED' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_AP_STAIPASSIGNED' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_AP_PROBEREQRECVED' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_GOT_IP6' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_ETH_START' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_ETH_STOP' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_ETH_CONNECTED' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_ETH_DISCONNECTED' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_ETH_GOT_IP' not handled in switch [-Wswitch]
src\main.cpp:135:12: warning: enumeration value 'SYSTEM_EVENT_MAX' not handled in switch [-Wswitch]
```

I also tried to use your precompiled binary. How to flash it to ESP32?

I also could switch to linux with some help wich programs/dependencies must be met...

Stefan

StofLE, 2018-09-08:
By the way: on Linux (Ubuntu and PlatformIO) similar output...

Stefan

sidoh, 2018-09-08:
I think the HTTP library being used made some breaking changes.  Should either update to stay compatible or find a way to fix to an older version.

You can flash the binary using [esptool](https://github.com/espressif/esptool).

closed 2018-09-20 by StofLE

joedeveloper, 2018-12-18:
If anyone else comes across this, I was able to move forward on osx by:
editing platformio.ini:

[common]
...
  ESP Async WebServer@1.1
....

I also removed lib44 with:
pio lib uninstall 44

then it was able to compile and run .. still having some issues getting the template settings or possibly pin wiring right

sidoh, 2018-12-18:
Thanks, @joedeveloper!  Would be happy to review/merge a PR. :)

joedeveloper, 2018-12-19:
I'd love to be able to offer one - but frankly it looks like it might be lib 44 that needs one.

>  also removed lib44 with:
pio lib uninstall 44
This was me misleading myself, the actual steps that resolved seems to have been what is mentioned [in platformio forums](https://community.platformio.org/t/errors-after-updating-espasyncwebserver/476)

in epaper_templates/.piolibdeps/Time_ID44/
rename Time.h to _Time.h - edit files in _that_ dir to reflect the file renaming.

Or we all just agree to use OS's with sane defaults. I am certainly missing it.

sidoh, 2018-12-19:
Ah yes, this exact issue has given me trouble in the past as well.  My usual fix for this kind of thing if the upstream project is inactive is to fork it and point my repo at a patched branch.