sidoh/epaper_templates #4: What's needed to expand support to other WaveShare e-paper display?

Frigorio, 2019-05-28:
Hi, luv your project. I've read that by now the only supported e-paper display is Waveshare 4"
Any chance any other will be son available? Can we help in any way?

sidoh, 2019-05-28:
All libraries supported by GxEPD2 are already supported in v1, at least theoretically.  I've only tested with the 2.9" display as it's the only one I have.

I can tag a dev release if you want a compiled binary, but everything should be checked into the v1.0.0 branch.

It's a little unpolished right now, the UI doesn't expose the setting yet, but you can set it via the REST API:

```
curl -vvv -X PUT -d '{"display.display_type":"GDEH029A1"}' -H 'Content-Type: application/json' http://epaper-display/settings
```

supported displays here:

https://github.com/sidoh/epaper_templates/blob/v1.0.0/lib/Display/DisplayTypeHelpers.cpp#L14

Frigorio, 2019-05-28:
Maybe GDEW027C44 is the one I need for this https://www.waveshare.com/2.7inch-e-paper-b.htm

sidoh, 2019-05-28:
Yeah, looks like it.  I've not tested it, though.  I've also not tested with any 3 color display.

sidoh, 2019-05-30:
Gonna close -- feel free to continue discussion if it's not working.

closed 2019-05-30 by sidoh

Frigorio, 2019-09-13:
After some time I managed to get that [Waveshare ESP32 e-paper board](https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board) up and running. 
**But** even if I've changed SPI pins according to manual and changed the display_type in settings nothing happens.
Here is my [test templates](https://github.com/sidoh/epaper_templates/files/3610690/templates.zip), my settings, my curl commands and related serial monitor:
```
{"display.display_type":"GDEW027C44",
"display.full_refresh_period":"16000",
"display.template_name":"/t/dashclock.json",
"hardware.busy_pin":"25",
"hardware.dc_pin":"27",
"hardware.rst_pin":"26"}

C:\>curl http://10.8.17.105/templates -X POST -F "dashclock=@dashclock.json"
{"success":true}
C:\>curl -H "Content-Type: application/json" http://10.8.17.105/settings -X PUT -d "{\"display.template_name\":\"/t/dashclock.json\"}"
{"success":true}

Applying settings
Loading new template: /t/dashclock.json
Loading template: /t/dashclock.json
Free heap - 206836
Formatted value: Wed 31.12.1969
Formatted value: 16:00
Formatted value: via Orfeo
Formatted value: Mercoledi
Rendering timestamp
Rendering timestamp
Rendering strade_pulite
Rendering raccolta_diff
_Update_Full : 9

C:\>curl http://10.8.17.105/templates -X POST -F "prova=@prova.json"
{"success":true}
C:\>curl -H "Content-Type: application/json" http://10.8.17.105/settings -X PUT -d "{\"display.template_name\":\"/t/prova.json\"}"
{"success":true}

Applying settings
Loading new template: /t/prova.json
Loading template: /t/prova.json
Free heap - 206592
_Update_Full : 9
```

sidoh, 2019-09-13:
Maybe try using these settings.  I think the new GxEPD library does not handle selection of the 
SPI bus 

[image]

I'd recommend verifying with a simple sketch (perhaps one from [GxEPD2](https://github.com/ZinggJM/GxEPD2/tree/master/examples)) that your hardware setup is working.

Frigorio, 2019-09-13:
> Maybe try using these settings. I think the new GxEPD library does not handle selection of the
> SPI bus
Nope... nothing happens.

> I'd recommend verifying with a simple sketch (perhaps one from [GxEPD2](https://github.com/ZinggJM/GxEPD2/tree/master/examples)) that your hardware setup is working.
My hardware setup is working. If I load the [default sketch](https://www.waveshare.com/wiki/File:E-Paper_ESP32_Driver_Board_Code.7z) from Waveshare it works flawlessly...

Frigorio, 2019-09-13:
[image]

sidoh, 2019-09-13:
Thanks for checking.

Can you try one from GxEPD2?  This is the library I'm using, and if it's not working, epaper_templates won't work either.

I'll be a bit slow on the response after this.  Display setup happens here in case you feel like poking around yourself:

https://github.com/sidoh/epaper_templates/blob/master/src/main.cpp#L52-L72