sidoh/epaper_templates #5: Expose setting to select SPI bus for ESP32

sidoh, 2019-05-28:


Ivanverdonck, 2020-03-15:
Hi, 

Where does this code need to be added?
I have a Waveshare ESP32 driver board. I tried adding this to the setup(), but this did not work.

I use a waveshare - GDEW075Z09     7.5" b/w/r
Is it possible to use the red color?

Kind regards,
Ivan

sidoh, 2020-03-19:
Hi Ivan,

If you're making the modification yourself, it should be enough to drop it somewhere in main.cpp (`initDisplay` is probably a good place https://github.com/sidoh/epaper_templates/blob/master/src/main.cpp#L54).

You might need to adjust the other pins as well (dc, rst, busy).

Unfortunately not currently possible to use the multi-color displays.

nullstalgia, 2020-03-29:
Notes:

```
SPI.end();
SPI.begin(13, 12, 14, 15); // replace these with the appropriate SPI pins
```
Can go both above and below the current `initDisplay()` code. [Example from the E-Paper driver lib.](https://github.com/ZinggJM/GxEPD2/blob/b2a40bac02df62d1a12cf3905773b2609a414a69/examples/GxEPD2_WS_ESP32_Driver/GxEPD2_WS_ESP32_Driver.ino#L88)

I personally do it above, but that's just me.

---

Also, this library *requires* hardware SPI busses. My tests with Software SPI ends up with the display doing nothing.

There is an internal fork for using Software SPI, but the author does not recommend that.

Luckily, the ESP32 has a spare (as shown by @sidoh above).

SPI1 isn't usually allowed to be used by users, as it's commonly used for SPI RAM (PSRAM), and SPI0 is the main system flash.

(If you wish, you can set up the PSRAM bus, but I would not recommend it)

As such, I put forth this suggestion for selecting the 2 free busses. This would fit best on the Hardware tab in the Settings page.

[image]

nullstalgia, 2020-03-30:
I started a little bit yesterday, but I got chumped when it came to the Web interface once again :sweat_smile: 

The main bit that matters is this:

```cpp
...
void initDisplay() {
  #if defined(ESP32)
    if(settings.hardware.spi_bus != HSPI){
        if (settings.hardware.spi_bus == VSPI){
            SPI.end();
            SPI.begin(18, 23, 19, 5);
        }
    }
  #endif
  if (display) {
    delete display;
    display = NULL;
  }
...
```
And in case you're not using an editor that shows the definitions of HSPI/VSPI easily, here are screenshots from VSCode

[image]
[image]

nullstalgia, 2020-03-30:
@sidoh, we should be good to close this if you feel the new features are satisfactory. :)

closed 2020-04-04 by sidoh