sidoh/epaper_templates #8: Unable to change settings via REST API

Frigorio, 2019-09-10:
I'm trying to change settings on my Waveshare ESP32 epaper board using REST API as described in [issue 4](https://github.com/sidoh/epaper_templates/issues/4#issuecomment-496566940) but nothing happens.
I've tried with both version 0.1.0 and 1.0.0 from the release page. No differences.

Here is the curl output:
```
C:\>curl -vvv -X PUT -d '{"display.display_type":"GDEW027C44"}' -H 'Content-Type: application/json' http://10.8.17.101/settings
* Could not resolve host: application
* Closing connection 0
curl: (6) Could not resolve host: application
*   Trying 10.8.17.101...
* TCP_NODELAY set
* Connected to 10.8.17.101 (10.8.17.101) port 80 (#1)
> PUT /settings HTTP/1.1
> Host: 10.8.17.101
> User-Agent: curl/7.55.1
> Accept: */*
> Content-Length: 35
> Content-Type: application/x-www-form-urlencoded
>
* upload completely sent off: 35 out of 35 bytes
* Empty reply from server
* Connection #1 to host 10.8.17.101 left intact
curl: (52) Empty reply from server
```
And here is the /settings page:
```
{"display.display_type":"GDEW042T2","display.full_refresh_period":"lu","display.template_name":"","hardware.busy_pin":"4","hardware.dc_pin":"0","hardware.rst_pin":"2","mqtt.password":"","mqtt.server":"","mqtt.username":"","mqtt.variables_topic_pattern":"","network.hostname":"epaper-display","network.mdns_name":"epaper-display","network.setup_ap_password":"waveshare","network.wifi_password":"############","network.wifi_ssid":"##########","system.timezone":"PT","web.admin_password":"","web.admin_username":"","web.port":"80"}
```
Any help?

sidoh, 2019-09-10:
Seems like your command isn't working like you want.  Maybe Windows doesn't like single ticks?  Haven't done CLI things on Windows in so long, so can't be much help there.  

But check out the errors near the top:

```
* Could not resolve host: application
* Closing connection 0
curl: (6) Could not resolve host: application
*   Trying 10.8.17.101...
```

also the content-type header is not getting properly set, which is the source of the problem:

```
> Content-Type: application/x-www-form-urlencoded
```

Frigorio, 2019-09-10:
Ok, thanks to [Stack Overflow](https://stackoverflow.com/a/12498151) now I have the correct syntax but still having a problem with the PUT command.
```
C:\>curl -vvv -H "Content-Type: application/json" -X PUT http://10.8.17.101/settings -d "{\"display.display_type\":\"GDEW027C44\"}"
*   Trying 10.8.17.101...
* TCP_NODELAY set
* Connected to 10.8.17.101 (10.8.17.101) port 80 (#0)
> PUT /settings HTTP/1.1
> Host: 10.8.17.101
> User-Agent: curl/7.55.1
> Accept: */*
> Content-Type: application/json
> Content-Length: 37
>
* upload completely sent off: 37 out of 37 bytes
< HTTP/1.1 400 Bad Request
< Content-Length: 2
< Content-Type: application/json
< Connection: close
< Accept-Ranges: none
<
{}* Closing connection 0
```

sidoh, 2019-09-10:
The same command is working for me, not sure what's going on.  

```
*   Trying 10.133.8.128...
* TCP_NODELAY set
* Connected to 10.133.8.128 (10.133.8.128) port 80 (#0)
> PUT /settings HTTP/1.1
> Host: 10.133.8.128
> User-Agent: curl/7.54.0
> Accept: */*
> Content-Type: application/json
> Content-Length: 36
>
* upload completely sent off: 36 out of 36 bytes
< HTTP/1.1 200 OK
< Content-Length: 16
< Content-Type: application/json
< Connection: close
< Accept-Ranges: none
<
* Closing connection 0
{"success":true}%
```

I assume you're on v1.0.0?

Does the UI not work for you?  Should be a place to enter this from there.

Frigorio, 2019-09-10:
> I assume you're on v1.0.0?
> Does the UI not work for you? Should be a place to enter this from there.

Yes, i'm on v1.0.0
Nope, the UI is not working for me. Here's a screenshot.
[image]

Frigorio, 2019-09-10:
I also have an ESP32 board but when I try to flash **epaper_templates_esp32-v1.0.0.bin**
 all I get is a boot-loop.
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
```

Frigorio, 2019-09-10:
I also have an ESP32 board but when I try to flash epaper_templates_esp32-v1.0.0.bin
all I get is a boot-loop.
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
```
Did I miss some important flag on esptool.py? 
Here is my command and output:
```
C:\>esptool.py --port COM9 write_flash --flash_size 4MB 0x00000 epaper_templates_esp32-v1.0.0.bin
esptool.py v2.6
Serial port COM9
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: cc:50:e3:bf:ae:88
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 1556288 bytes to 1018895...
Wrote 1556288 bytes (1018895 compressed) at 0x00000000 in 90.0 seconds (effective 138.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

sidoh, 2019-09-11:
Ah.. you know, I haven't tested with an ESP8266 in a while.  Totally possible that it's either broken or FUBAR.  I switched to ESP32 after running too close to the wire with RAM constraints on the ESP8266.  Having a 400x300 bit buffer in memory isn't awesome when you've only got 20 KB of memory to play with 😁.  I'll give it a go now.

The issue with ESP32 is almost certainly partition table related.  ESP32s have [partition tables](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/partition-tables.html) which enable multiple apps on the same chip and a bunch of other fancy stuff.  Because the bundled UI is so enormous, I'm using a non-default partition table which gives more room to the app and less to SPIFFS.

Unfortunately it's pretty difficult to change the partition table with esptool.py.  It's possible, but I've never tried.  ESP-IDF has [a tool](https://github.com/espressif/esp-idf/issues/1494) that wraps esptool.py to accomplish this.

But probably the easiest way to get going here is to just give up on the pre-compiled binary and flash with PlatformIO.  After installing (`pip install platformio` should do the trick), check out epaper_templates source and run this command:

```
pio run -e esp32 --target upload
```

This will take care of the partition table crap and upload the image for you.  After this, flashing with esptool.py or using OTA will work because the partition table has been set properly.

sidoh, 2019-09-11:
I pushed a fix for the webpage, and confirmed saving settings works on an ESP32.

Frigorio, 2019-09-13:
> But probably the easiest way to get going here is to just give up on the pre-compiled binary and flash with PlatformIO. After installing (`pip install platformio` should do the trick), check out epaper_templates source and run this command:
> ```
> pio run -e esp32 --target upload
> ```
Ok, I've started a Debian machine and everything is ok now (damn Wizoz).
I can confirm that saving settings via curl works.
```
C:\>curl -v -H "Content-Type: application/json" http://10.8.17.105/settings -X PUT -d "{\"display.display_type\":\"GDEW027C44\"}"
*   Trying 10.8.17.105...
* TCP_NODELAY set
* Connected to 10.8.17.105 (10.8.17.105) port 80 (#0)
> PUT /settings HTTP/1.1
> Host: 10.8.17.105
> User-Agent: curl/7.55.1
> Accept: */*
> Content-Type: application/json
> Content-Length: 37
>
* upload completely sent off: 37 out of 37 bytes
< HTTP/1.1 200 OK
< Content-Length: 16
< Content-Type: application/json
< Connection: close
< Accept-Ranges: none
<
{"success":true}* Closing connection 0
```
The webpage is not working for me (same output as [post above](https://github.com/sidoh/epaper_templates/issues/8#issuecomment-530011160)), but not a big problem, i'll go with the curl/mqtt way. My version is:
`{"version":"v1.0.1","variant":"unknown","free_heap":203184,"sdk_version":"v3.2-18-g977854975"}`

But there is a thing I'm not able to modify: the timezone. When trying to send something different from PT the serial monitor says:
`WARN - couldn't find specified timezone.  Returning default.`
and nothing happens. Any idea?

closed 2019-09-13 by Frigorio

sidoh, 2019-09-13:
You might need to force-refresh the webpage for it to work.  Some old assets might be cached.

These are the only timezones I've got right now.  Easy to add more:

https://github.com/sidoh/epaper_templates/blob/master/lib/Time/Timezones.cpp#L34

Feel free to open a PR, of course =)