sidoh/epaper_templates #12: Not able to compile with PlatformIO (Or upload via ESPTool)

nullstalgia, 2020-03-21:
Heyo.

So first, ESPTool.

I've been running `esptool.py write_flash X epaper...`, with X being 0x0, 0x1000, and 0x10000. None have been working, causing this error:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:571968
```
, which might be due to incorrect DIO/DO/DI setting? Not sure.

---

And with PlatformIO, I have:

```
$ node --version
v10.19.0
$ npm --version
6.14.2
```

After running `pio lib install` to get all dependencies, it fails at building the web pages.

```
Attempting to build webpage...
npm WARN @neutrinojs/style-minify@8.3.0 requires a peer of neutrino@^8.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN bootstrap@4.3.1 requires a peer of jquery@1.9.1 - 3 but none is installed. You must install peer dependencies yourself.
npm WARN eslint-plugin-react@7.16.0 requires a peer of eslint@^3.0.0 || ^4.0.0 || ^5.0.0 || ^6.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN eslint-plugin-react-hooks@2.2.0 requires a peer of eslint@^3.0.0 || ^4.0.0 || ^5.0.0 || ^6.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN epaper_web3@1.0.0 No description
npm WARN epaper_web3@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

b'audited 13128 packages in 4.901s\n\n2 packages are looking for funding\n  run `npm fund` for details\n\nfound 369 vulnerabilities (368 low, 1 moderate)\n  run `npm audit fix` to fix them, or `npm audit` for details\n'
(node:14965) DeprecationWarning: Tapable.plugin is deprecated. Use new API on `.hooks` instead
b'\n> epaper_web3@1.0.0 build /home/tony/Documents/PlatformIO/Projects/epaper_templates-master/web\n> webpack --mode production\n\nHash: 85bea136ff2616102151\nVersion: webpack 4.41.2\nTime: 6874ms\nBuilt at: 03/20/2020 7:36:09 PM\n                        Asset       Size  Chunks                                Chunk Names\n         assets/2.8b74afbd.js    748 KiB       2  [emitted] [immutable]  [big]  \n      assets/2.8b74afbd.js.gz    212 KiB          [emitted]                     \n     assets/index.63137acc.js   90.1 KiB       0  [emitted] [immutable]         index\n  assets/index.63137acc.js.gz   23.2 KiB          [emitted]                     \n    assets/index.d5272d06.css    149 KiB       0  [emitted] [immutable]         index\n assets/index.d5272d06.css.gz   24.1 KiB          [emitted]                     \n   assets/runtime.07e3d730.js   1.46 KiB       1  [emitted] [immutable]         runtime\nassets/runtime.07e3d730.js.gz  724 bytes          [emitted]                     \n                   index.html  406 bytes          [emitted]                     \n                index.html.gz  261 bytes          [emitted]                     \n                 web_assets.h    930 KiB          [emitted]              [big]  \n\nWARNING in asset size limit: The following asset(s) exceed the recommended size limit (244 KiB).\nThis can impact web performance.\nAssets: \n  assets/2.8b74afbd.js (748 KiB)\n  web_assets.h (930 KiB)\n\nWARNING in entrypoint size limit: The following entrypoint(s) combined asset size exceeds the recommended limit (244 KiB). This can impact web performance.\nEntrypoints:\n  index (989 KiB)\n      assets/runtime.07e3d730.js\n      assets/2.8b74afbd.js\n      assets/index.d5272d06.css\n      assets/index.63137acc.js\n\n\nWARNING in webpack performance recommendations: \nYou can limit the size of your bundles by using import() or require.ensure to lazy load some parts of your application.\nFor more info visit https://webpack.js.org/guides/code-splitting/\n'
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.9.0 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 2.10002.190628 (1.0.2) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File ".get_version.py", line 31, in <module>
    sys.stdout.write("-DEPAPER_TEMPLATES_VERSION=%s %s" % (version.decode('utf-8'), ' '.join(sys.argv[1:])))
AttributeError: 'str' object has no attribute 'decode'
OSError: 'python3 .get_version.py -DMQTT_DEBUG -Idist' exited 1:
  File "/home/tony/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 156:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/tony/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/tony/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/tony/.platformio/platforms/espressif32@1.9.0/builder/main.py", line 209:
    target_elf = env.BuildProgram()
  File "/home/tony/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/tony/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 61:
    env.ProcessProgramDeps()
  File "/home/tony/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/tony/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 115:
    env.ProcessFlags(env.get("BUILD_FLAGS"))
  File "/home/tony/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/tony/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 213:
    env.Append(**env.ParseFlagsExtended(flags))
  File "/home/tony/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/tony/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 176:
    for key, value in env.ParseFlags(str(raw)).items():
  File "/home/tony/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 805:
    do_parse(arg)
  File "/home/tony/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 671:
    arg = self.backtick(arg[1:])
  File "/home/tony/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 589:
    raise OSError("'%s' exited %d" % (command, status))
```

And even after making a package.json with `npm init` (or without it), installing all of the peer deps with `install-peerdep` or manually, it's giving the same error. :c

sidoh, 2020-03-21:
Hello!

Thanks for sharing the errors, and appreciate how many things you already tried on your own :)

I believe the first message is a normal part of the boot process.  If it's hanging indefinitely, try unplugging everything except UART and rebooting.  Some GPIOs affect the boot process.

Here's the command I've used to flash the pre-compiled binary in case it's helpful:

```
esptool.py -b 460800 write_flash 0x10000 epaper_templates_esp32-v1.0.0.bin
```

The other one looks like the script I use to tag the version (`.get_version.py`) is barfing because it doesn't think the project is a git repo.  Did you by chance download the project as a zip or some other way?  If so, could you try checking it out via git?

```
git clone https://github.com/sidoh/epaper_templates.git
```

should do the trick.  Let me know if that doesn't work and I can push a fix to make it fall back to a saner default behavior.

nullstalgia, 2020-03-23:
With ESPTool, that was with 2 different boards, each with all GPIOs disconnected, with it bootlooping with that message. I have a theory, however.

On one of your boards, run a `esptool.py erase_flash` and then upload just that binary.

It'll upload correctly, but will fail to boot.

When uploading via PlatformIO, it does more than just that binary,

```
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 16848 bytes to 10928...

Writing at 0x00001000... (100 %)
Wrote 16848 bytes (10928 compressed) at 0x00001000 in 0.2 seconds (effective 545.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.0 seconds (effective 3512.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 13142.0 kbit/s)...
Hash of data verified.
Compressed 1654400 bytes to 1011185...

Writing at 0x00010000... (1 %)
Writing at 0x00014000... (3 %)
Writing at 0x00018000... (4 %)
Writing at 0x0001c000... (6 %)
Writing at 0x00020000... (8 %)
Writing at 0x00024000... (9 %)
```

And even trying to upload a sketch via Arduino, then uploading the epaper firmware...

Or a PlatformIO sketch, then uploading the epaper firmware...

~~It ends up with similar issues. So it appears the best way to get the initial sketch onto the board, is to upload via PlatformIO. Then you can likely supply that .bin to the firmware updater. :)~~

Looking through the issues while writing this, it looks like #8 had the same problem, and you guys came across a similar solution.

---

And with my testing, I have found a few issues I would like to help resolve! 

As long as you're okay with it, I'll be submitting some issues (and if I get comfortable enough with the project, some pull requests). Because I love to help smaller projects grow and become easier to use for both new people, and people who want to get stuff done quick.

I do have a question, though. Would you be okay if I make an feature request-issue for each idea that seems feasible? As I have a few topics I could submit right now.

(Deep Sleep, guides for adding fonts and timezones, more descriptive examples, probably a wiki for all of this information, etc.)

sidoh, 2020-03-23:
Thanks for the update!

Ah, there is more that flashing with pio does.  It sets up the [partition table](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html).  I had forgotten that I needed to change that to accommodate the large web assets.  I believe there's a way to update the partition table with esptool, but haven't bothered to look into it.  Definitely worth updating the docs.

Of course, happy to collaborate!  Feel free to submit issues/PRs in whatever way feels most natural to you.

nullstalgia, 2020-03-23:
Well I did try to send in that table as well via ESPTool (0x8000, i think is the address. It's generated next to firmware.bin), but that didn't change anything.

sidoh, 2020-03-23:
Gotcha, interesting.  Will have to play around more with pio to see what it's doing.

insaneFu, 2020-04-01:
Hello there,

first of all, cheers to the project idea, it looks very promising to me and I'd very much like to use it, especially for its MQTT capabilities (running a nodered based smart home...) and of course the template idea itself.

However, I also cannot flash via esptool.py nor the ESP32 Download Tool, for obvious reasons. (Missing altered partition table as far as I unterstand.) Though I don't get any errors while uploading, I don't get to the point where I can join the setup AP to configure. Including the mentioned 0x10000 firmware offset upon flashing didn't change a thing.

I must confess, don't even know where to look for errors, since I am really new to ESP development. Didn't get the Visual Studio/platform IO project to build/upload either.

I will be using the Azdelivery nodemcu esp32 and a Waveshare 7.5" HAT v2 b/w , and although I probably won't be much help in development, I'd be more than happy to test and report back - any help or advice will be greatly appreciated!

sidoh, 2020-04-01:
[This post](https://community.platformio.org/t/export-of-binary-firmware-files-for-esp32-download-tool/9253) looks helpful.  Maybe a way to combine everything into a mega-bin.  I'll try it out sometime soon.

sidoh, 2020-04-01:
For now, uploading with platformio is the easiest way to get going.  Very easy to install with pip:

```
pip3 install platformio
python3 -m platformio run -e esp32 --target upload
```

nullstalgia, 2020-04-02:
I could have sworn I mentioned that somewhere but I can't find it

You could just do a `esptool.py read_flash...` with the full address range instead of using a tool like that. :)

sidoh, 2020-04-02:
You probably communicated the same information, even if not the link.  Sorry I missed it.

The images right now are built and uploaded by Travis CI, which is really only capable of running the `compile` target.

nullstalgia, 2020-04-02:
Nah, wasn't the link, just the concept. Sorry for not being as clear.

When I'm back home I'll try to see if I can break apart what esptool does when uploading via platformio.

I'm sure it's just something simple that we're missing for the initial partitioning. Something we could supply in the release / repo somewhere.

But can't you upload a custom file for releases? That's what I've done with some private repos

sidoh, 2020-04-02:
yeah, from the article, looks like these are the bits --

```
0x1000 /Users/me/.platformio/packages/framework-arduinoespressif32/tools/sdk/bin/bootloader_dio_40m.bin
0x8000 /Users/me/Documents/PlatformIO/Projects/ESP32Proj/.pio/build/esp32doit-devkit-v1/partitions.bin
0xe000 /Users/me/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin
0x10000 .pio/build/esp32doit-devkit-v1/firmware.bin
```

* Bootloader
* Partition table
* ?`boot_app0.bin`?
* Firmware

Guessing the last three are the necessary things.

nullstalgia, 2020-04-02:
Hmm
I worry about the bootloader, to be honest. Since I'm not sure if all 32's
ship with 4mb.

But from what I've seen, most boards ship with a WROOM module. So if we can
base it off of that, I think we would be in the clear.

sidoh, 2020-04-02:
Makes sense.

The way I handle this for my other projects is to build and upload multiple binaries for each supported platform.  Would be easy to set up here if that ends up being the case.

I didn't answer your question from before, sorry:

> But can't you upload a custom file for releases? That's what I've done with some private repos

You definitely can, but I really like for it to be an automatic part of the release process.  Right now every tagged version results in a release with pre-built binaries.

nullstalgia, 2020-04-02:
Hmmm

Well, if it's anything like I think it is,

(Let's be frank, might not be the case but would be good)

you could have a few bootloaders and partition tables in a few folders, it
might be possible to concatenate them together with a python script at the
buildbot.

That way, you have multiple binaries for the platforms, it's automated, and
can work from a 0'd flash chip.

sidoh, 2020-04-03:
It does not feel like there's going to be a terribly elegant way to handle this, but reporting progress anyway:

The espressif32 PlatformIO platform [always uploads all four of these things](https://github.com/platformio/platform-espressif32/blob/f923797a3066c1cf55f6a8679484ae7953f9e3c9/builder/main.py#L336) with `--target upload`.

I verified that a mega-bin containing all four parts can be uploaded in a single go. (cobbled together with a hacky script)

I'm 99% sure I see how to create the mega-bin as a part of the PlatformIO build.  I've seen what I need to to believe that this is possible. 

The remaining concerns are:

* Haven't fully thought this through, but there'd probably have to be _both_ a "full" and "update" version of each release.  OTA updates wouldn't work with the full version.
* In this model, uploading the full bin instead of the update would wreck the install.

Likely possible to alleviate these problems with good input validation, but not sure.

insaneFu, 2020-04-03:
I have tried building the project with platformIO via VSCode, via Atom, and on the pure PowerShell. Sadly all without success (missing packets which I could partly resolve but then also with errors I cannot troubleshoot):

`npm ERR! syscall access
npm ERR! path N:\_Downloads\PinkHome\epaper_templates\web\node_modules\are-we-there-yet\node_modules\readable-stream
npm ERR! errno -4058
npm ERR! enoent ENOENT: no such file or directory, access 'N:\_Downloads\PinkHome\epaper_templates\web\node_modules\are-we-there-yet\node_modules\readable-stream'
npm ERR! enoent This is related to npm not being able to find a file.

npm ERR! enoent 
 
npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\-----\AppData\Roaming\npm-cache\_logs\2020-04-03T13_42_08_092Z-debug.log
TypeError: can only concatenate str (not "CalledProcessError") to str:
  File "C:\Users\-----\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 154:
    env.SConscript(item, exports="env")
  File "C:\Users\-----\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\-----\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "N:\_Downloads\PinkHome\epaper_templates\.build_web.py", line 77:
    build_web()
  File "N:\_Downloads\PinkHome\epaper_templates\.build_web.py", line 36:
    raise BaseException("Error building web assets: " + e)
========================== [FAILED] Took 5.71 seconds ==========================`

But since I am still trying all of that on a Windows 10 machine, I hope I will soon find the time to setup one of my old laptops with a linux distro soon and try to build on there again.

Will also gladly try the mega-bin when available. Thanks guys, you're awesome!

sidoh, 2020-04-03:
Attaching the cobbled-together mega-bin.  Can't yet promise these will be part of the release process, but this one should work.

[mega_bin.bin.zip](https://github.com/sidoh/epaper_templates/files/4428254/mega_bin.bin.zip)

sidoh, 2020-04-03:
Got a script that builds the full binary working:

```python
#!/usr/bin/python3

from pathlib import Path

Import("env")

SOURCE_START_LOCATION = 0x10000

def create_full_bin(source, target, env):
    firmware_file = target[0].get_abspath()
    full_image_filename = env.subst("$BUILD_DIR/${PROGNAME}-full.bin")
    parts = []

    # Will contain 3 parts outside of the main firmware image:
    #   * Bootloader
    #   * Compiled partition table
    #   * App loader
    for part in env.get("FLASH_EXTRA_IMAGES", []):
        start, filename = part

        # Parse hext string
        if isinstance(start, str):
            start = int(start, 16)

        filename = env.subst(filename)

        parts.append((start, filename))

    # End with the main firmware image
    parts.append((SOURCE_START_LOCATION, firmware_file))

    # Start at location of earliest image (don't start at 0x0)
    ix = parts[0][0]

    with open(full_image_filename, "wb") as f:
        while len(parts) > 0:
            part_ix, part_filename = parts.pop(0)
            part_filesize = Path(part_filename).stat().st_size
            padding = part_ix - ix

            for _ in range(padding):
                f.write(b"\x00")

            with open(part_filename, "rb") as part_file:
                f.write(part_file.read())

            ix = part_ix + part_filesize

env.AddPostAction("$BUILD_DIR/${PROGNAME}.bin", create_full_bin)
```

needs a patch to platformio.ini --

```diff
diff --git a/platformio.ini b/platformio.ini
index ec9c3d4..ff78372 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -40,6 +40,7 @@ lib_deps_external =
   GxEPD2@~1.2.3
 extra_scripts =
   pre:.build_web.py
+  post:.build_full_image.py
```

nullstalgia, 2020-04-03:
I've seen something similar done before where the user uploads a kind of "blank-slate" bin that is just a prepper for the actual firmware bin to be uploaded via the blank slate. The blank-slate does nothing but host the firmware installer, and will install the proper firmware once uploaded.

insaneFu, 2020-04-04:
At which start address would I flash that mega bin?

nullstalgia, 2020-04-04:
@insaneFu 

~~I would guess 0x0~~

0x1000 is confirmed working on my end.

unicornrobot, 2020-04-04:
mega bin uploaded successfully. Changed settings>display type to my display (GDE0213B1)
Created a test template but screen is not updating on save/activate. Am I missing something?
updated firmware to 2.3.0 too.

insaneFu, 2020-04-04:
@nullstalgia @sidoh 

0x1000 confirmed. Got the mega_bin flashed succesfully (in Windows with the flash download tool), WiFi config AP worked and device got on my own network just now. :) Still need to hook up my 7.5" display, though. And try to configure something sensible.

Right in time my linux machine was finished installing and now I don't need it anymore^^ - for now, at least. We'll see.

Need to repeat: you're both awesome. Thank you so much!

So, I'll be back for more later. Cheers.

nullstalgia, 2020-04-04:
@unicornrobot Confirm you set up the extra pins in the Hardware tab, and that you follow the correct SPI pinout according to the README.

insaneFu, 2020-04-04:
So I succesfully setup the device in my wifi and it already posted its status to my MQTT server. :)

However, the display itself doesn't change pictures. Doesn't even refresh or anything. It's like its not even connected, still displaying the content I had set before with the raspi.

I am pretty shure I connected all the pins correctly (according to readme plus changed the custom pins to free ones because my wroom esp32 was using the preset custom pins). Chose the right display size and resolution, lowered the full refresh period to 60000, so I should have seen changes within a minute. Also changed the config switch on the pi HAT controller  from 4pin SPI to 3pin SPI.(Was that correct?)

Setup a little test template with some lines, rectangles, a bitmap and the preconfigured time and wifi variables. Didn't really get what the formatters prompt upon template creation was about, but activated nonetheless and the device actually stored the template persisting reboots.

But as stated, the display itself doesn't print. :( Any ideas? Oh, and also my time zone (CET) is not in the list, but that is something I can live with for now... ;)

nullstalgia, 2020-04-04:
@insaneFu Thanks for the detailed description!

But you actually want 4-pin SPI.

And as for timezones, I'm planning on having a guide of sorts for adding new timezones. Since CET isn't that much of a "niche" timezone, I don't think @sidoh would be against it being pushed to the master branch.

sidoh, 2020-04-04:
@insaneFu, if things are working properly, you'll see a refresh immediately after boot or settings saved (including saving a template).  Sounds like a hw issue if you're not seeing any response from the screen.

The only reason the list of timezone selections is so sparse is because I'm lazy.  I'd like for most major timezones to be there, just haven't gotten around to it.  Happy to add yours as a oneoff, but there should definitely be a broader selection.

sidoh, 2020-04-04:
Calling the megabins "initializers" in an attempt to be slightly more descriptive.

README is updated with instructions, and the [v2.4.0 release](Add link to Github project in Web UI) has an initializer binary.

Gonna close since I think this is resolved, but let me know if not.

@insaneFu, if you're still having hardware setup issues, feel free to open another ticket.

closed 2020-04-04 by sidoh

insaneFu, 2020-04-04:
First time I conciously saw the reaction emojis, thanks, haha. :)

I rewired my display to the raspi, this time not using the HAT but the breakout cables for pin connections right to the raspi GPIOs, that way confirmed that the display itself and the breakout cables from the controller board are still working fine.

In the meantime I successfully bricked the esp32 flash by overloading the memory with a full 800x480 jpeg upload, but thankfully the erase and reflash instantly worked this time as expected... ;) Maybe in the long term a kind of memory overload protection would be nice.

4-line SPI confirmed, thanks for that. After the test I connected the display to the esp32 again, this time choosing different custom pins. Still no display update. I am stuck. Any more ideas? Should I try building and flashing from source on my linux machine?