sidoh/epaper_templates #13: Add link to GxEPD2/Add descriptors to Display Type dropdown

nullstalgia, 2020-03-23:
So this was my first major once I got the software loaded onto the board. Currently, the dropdown list for the display types is just model numbers. And even looking on the linked Waveshare wiki to find my display (the default 4.2" B/W), I wasn't able to find any thing like that in the datasheet or manual!

I eventually opened up the GxEPD2 repo, and found the list of displays, and I was able to nail down which one was mine, which ended up being the default.

Personally, I'm okay with keeping the long model numbers there, but having either a link or a similar summary as the one on the GxEPD2 repo would be ideal.

[image]

[image]

I personally prefer the latter as it doesn't require going to another site, especially if outside internet connectivity is not possible at that moment.

sidoh, 2020-03-23:
Makes sense, great idea.

Should be a way to serve it in the [`/api/v1/screens`](https://github.com/sidoh/epaper_templates/blob/master/lib/HTTP/EpaperWebServer.cpp#L584) route.  The list is in [DisplayTypeHelpers.cpp](https://github.com/sidoh/epaper_templates/blob/master/lib/Display/DisplayTypeHelpers.cpp).

nullstalgia, 2020-03-29:
Closing due to pull #14 fixing this.

closed 2020-03-29 by nullstalgia