sidoh/epaper_templates #17: MQTT Implimentation is unstable

nullstalgia, 2020-03-24:
I'm not entirely sure if this is due to your implementation of the library, or the library itself. I can say, however, that I did try the latest version of the library (last "release" of library was in 2018), and the following issues persisted.

Issues I noticed:

1. Doesn't always connect

    - Not as common as the other issues

2. Saving a template loses connection to broker

3. When the MQTT Broker restarts, it fails to reconnect

    - In the Serial output, it just says "connected disconnected connected disconnected..."

4. The custom client ID (similar to the custom AP SSID) fails to register

    - When setting it via a string directly like
    - `mqttClient.setClientId("big_potato");` it works
    - But no matter how I tried to copy the custom string to a char then give it to the function, my broker always spat out an auto-generated client ID... maybe it wants a pointer? Will have to try that later

My broker is:

Mosquitto (latest as of March 23, 2020), running in Docker on a Raspi 3

sidoh, 2020-03-25:
re: (1) and (3), which seem probably related -- interesting.  I definitely intended for automatic reconnections to happen, and I did at some point test that it worked, but perhaps something about it broke.  I'll look into it.

(2) is sort of expected.  Saving a template involves updating settings, and when settings are updated, everything is reloaded.  It's possible to do smart diffing to prevent unchanged parts from getting reloaded, but obviously involves some moderately fancy ~liability~ code to accomplish.

Will look into (4), can confirm I see this as well.  Probably the case that the client library wants you to manage the memory (which is what I think you're suggesting).  Although I'd expect to sometimes see a garbage client name instead of null if this were the case...

sidoh, 2020-03-25:
I think https://github.com/sidoh/epaper_templates/commit/089597a3e2385995fdd107704c108d202897c1aa should fix (4).  Makes sense that this was an issue given [what `setClientID` does](https://github.com/marvinroger/async-mqtt-client/blob/master/src/AsyncMqttClient.cpp#L63).

log from after the patch --

```
1585112780: New client connected from 10.133.8.108 as epaper-display-312782384 (...)
```

sidoh, 2020-03-25:
This is pretty amusing.  I think (1), (3), and (4) are all the same problem.  Before fixing (4), I saw the connected/disconnected loop a couple of times.  Now I'm not seeing it at all, even after forcing a reconnect 20+ times.

I think what was happening is the client ID memory was getting reclaimed, overwritten, and the MQTT client library (understandably) didn't know how to deal.  Was probably sending invalid packets to the MQTT server, which caused disconnects.

There was an obnoxious intermittent problem with the two displays I had where they'd drop their MQTT connection and fail to reconnect unless I restarted them.  I think this is why.  Thanks so much for reporting!

sidoh, 2020-03-25:
Should be fixed in [v2.1.4](https://github.com/sidoh/epaper_templates/releases/tag/v2.1.4).

nullstalgia, 2020-03-25:
Confirmed (3) and (4) fixed.

(1) testing will take time.

(2) ~~is up to you. But a stopgap/bandaid solution would be to have a note in the MQTT page saying that a restart is needed after changing the template for MQTT to function again.~~

Actually, I think (2) was also fixed by (4)! Pesky little (4).

sidoh, 2020-03-25:
Gotcha, so the problem was that the MQTT connection didn't come back after saving settings, not that it reconnects at all.

Happy to leave it open for a few days while we make sure it's stabilized, but sounds like we're expecting all of the issues are fixed.

nullstalgia, 2020-03-29:
Closing as neither of us have seen issues so far, and the release seems to have gone smoothly. :)

closed 2020-03-29 by nullstalgia