sidoh/epaper_templates #27: Bitmap dimensions that aren't a power of 2 are not rendered correctly.

nullstalgia, 2020-04-04:
20x20:

[image]

32x32:

[image]

sidoh, 2020-04-04:
Thanks, was able to reproduce on my end.

I think AdafruitGFX might be incapable of rendering bitmaps that don't have a width divisible by 8 ([see code](https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_GFX.cpp#L790)).  Having trouble thinking of a good reason for this (it's honestly a pretty goofy restriction), but that's how it is.

Confirmation: a 20x20 image doesn't work, but a 24x24 does.

As goofy as this feels, the best solution is probably just to re-implement `renderBitmap` in a way that supports arbitrary widths.  This isn't hard, and I don't quite understand why AdafruitGFX works the way it does.

sidoh, 2020-04-04:
Should be fixed in #28.

nullstalgia, 2020-04-04:
Will test very soon, thank you!

---

Unrelated, but are you going to close #5 since it has been solved?

nullstalgia, 2020-04-04:
#28 does fix it.

But I have to ask, why are we able to change the width and height if doing anything but the actual size of the bitmap? Wouldn't it be best to set it from the given values in the Metadata? I'll write that out as and send it as a pull and you can see if you like it or not.

sidoh, 2020-04-04:
Great, thanks for testing.

The bitmap metadata is relatively new, and was developed in the context of UI hinting.  The backend could definitely take advantage of the metadata in a way that it isn't right now.

closed 2020-04-04 by sidoh