sidoh/epaper_templates #32: Display not updating

insaneFu, 2020-04-04:
First time I conciously saw the reaction emojis, thanks, haha. :)

I rewired my display to the raspi, this time not using the HAT but the breakout cables for pin connections right to the raspi GPIOs, that way confirmed that the display itself and the breakout cables from the controller board are still working fine.

In the meantime I successfully bricked the esp32 flash by overloading the memory with a full 800x480 jpeg upload, but thankfully the erase and reflash instantly worked this time as expected... ;) Maybe in the long term a kind of memory overload protection would be nice.

4-line SPI confirmed, thanks for that. After the test I connected the display to the esp32 again, this time choosing different custom pins. Still no display update. I am stuck. Any more ideas? Should I try building and flashing from source on my linux machine?

sidoh, 2020-04-04:
Oops!  Can probably do some kind of validation on uploads in the future, but it's pretty tricky to make completely bulletproof.

Are you able to attach some pictures of your setup?

You might also want to try setting up one of the [GxEPD2 examples](https://github.com/ZinggJM/GxEPD2/tree/master/examples).  If you can get it working with this, there's a problem with epaper_templates.

nullstalgia, 2020-04-04:
+1 On hard to make bulletproof. I used to do basic PHP work and sometimes you couldn't be sure that the file size you were given was 

1. Accurate

2. Actually there

But yeah, if you could send pictures/a vid of your set up, that would be great. In addition, sending the result of `<ip_address>/api/v1/settings` could help us make sure everything is saved and configured properly.

(You can use
\`\`\`json
\<data\> 
\`\`\`
to format it nicely)

## Make sure to remove your WiFi Password! :laughing:

nullstalgia, 2020-04-04:
@sidoh Might be smart to make an api call that outputs a **privacy-safe** file for when people submit issues.

Could include current free heap, selected display, pins, etc... but strips out sensitive info such as passwords and IP addresses.

`/api/v1/report` ? :)

insaneFu, 2020-04-05:
[image]
[image]
[image]

```json<data>
{
   "display.display_type":"GDEW075T7",
   "display.full_refresh_period":"60000",
   "display.template_name":"/t/test.json",
   "display.windowed_updates":"false",
   "hardware.busy_pin":"5",
   "hardware.dc_pin":"19",
   "hardware.rst_pin":"18",
   "hardware.spi_bus":"HSPI",
   "mqtt.client_status_topic":"stats/epaper/tele",
   "mqtt.password":"",
   "mqtt.server":"192.168.23.2",
   "mqtt.username":"",
   "mqtt.variables_topic_pattern":"",
   "network.hostname":"epaper-display",
   "network.mdns_name":"epaper-display",
   "network.setup_ap_password":"waveshare",
   "network.wifi_password":"",
   "network.wifi_ssid":"",
   "power.awake_duration":"30",
   "power.sleep_duration":"600",
   "power.sleep_mode":"ALWAYS_ON",
   "power.sleep_override_pin":"25",
   "power.sleep_override_value":"1",
   "system.timezone":"UK",
   "web.admin_password":"",
   "web.admin_username":"",
   "web.port":"80"
}
```

So, there are pictures of my hardware setup for you as well as my settings json. My MQTT server doesn't use credentials, so nothing to hide there. Funny thing about the wifi-config though is that it connects to my own wifi correctly as configured via the setup AP but both here and in the web interface (same data I guess), the default wifi config is still displayed.

Since there is a new mega bin version with my timezone (thank you! 👍 ) now, first thing I'll try is flash that and see if there is a difference. After that, I guess I'll try the [examples](https://github.com/ZinggJM/GxEPD2/tree/master/examples)  and/or build from source after all.

insaneFu, 2020-04-05:
@sidoh @nullstalgia 
Finally stuff is happening as intended! :) I had worries about the display being the V2 version with 800x480 and the driver being incompatible, then in the GxEPD2 description it was mentioned that 

> note that 7.5" e-paper displays don't work reliable if fed from 3.3V Arduino pin

and I wondered whether that might also be the case with my esp32.

But in the end it seems it was just a matter of changing the pins to VSPI (see new picture) and of course reconfiguring those in the firmware, and I think also important was chosing a different ground pin (between G12 and G13 on my board, and there is only one 3V3 right at the end, which works just fine).

[image]

So, in the end, building the firmware by myself was not even necessary and now I can finally move on to the config. Yay!

Thanks again, you two. Very pleased by your willingness to help! Keep on!

closed 2020-04-05 by insaneFu