sidoh/epaper_templates #35: Force a reboot after setting up deep sleep (or allow button to be pushed when sleeping)

nullstalgia, 2020-04-05:
So I finally set up my ESP32 Corona panel, but I was unable to start the deep sleep cycle until I used a wire to bridge EN to GND because of this command not working on either of my ESP32's

`curl -d "command=reboot" -X POST http://192.168.86.45/api/v1/system` (One didn't have deep sleep enabled)

And the Reboot button was disabled. Enabling it via Inspect Element didn't help.

sidoh, 2020-04-05:
The reboot button in the UI is only disabled when it thinks the system is already rebooting (it detects this by a regular poll to the `system` endpoint).

I wouldn't expect that curl command to work.  The API only supports JSON inputs.  This is probably what you want:

```
 curl -v -d '{"command":"reboot"}' -H'Content-Type: application/json' 10.133.8.108/api/v1/system
```

If this doesn't work to enable deep sleep, it's possible that a software reset never works.

nullstalgia, 2020-04-05:
That command did seem to work! I didn't think that it would also be in a JSON format :P

closed 2020-04-05 by sidoh