sidoh/epaper_templates #41: No Display using Waveshare esp32 driverboard

xx-robot-xx, 2020-07-08:
https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board

This board uses a different set of spi pins.
(https://github.com/ZinggJM/GxEPD2/tree/master/examples/GxEPD2_WS_ESP32_Driver)

sck-13
din-14
cs-15
busy-25
rst-26
dc-27

I am using this board with a 7.5inch b/w/r display, and when i save, i get no output (i use hspi with configurable pins set correctly)

Is it possible to change the  spi pins for sck, din , cs ?

sidoh, 2020-07-08:
There's a setting to change the SPI bus under the "hardware" tab:

[image]

See the README for more information -- https://github.com/sidoh/epaper_templates#spi-bus-and-pin-selection

Does this do what you need it to?

xx-robot-xx, 2020-07-08:
Yes, I configured like mentioned in the readme

In that page i configured the following spi info

**Busy Pin** 25
**DC Pin** 27
**RST Pin** 26

However, in the README, the rest of the spi :

                           HSPI (default)
DI (MOSI)         12
CLK                  14
CS (SS)            15

But the waveshare driver board has,

DI - 14
CLK - 13
CS - 15

as mentioned here https://github.com/ZinggJM/GxEPD2/blob/master/examples/GxEPD2_WS_ESP32_Driver/GxEPD2_WS_ESP32_Driver.ino

But the webpage doesnt allow changes to those parameters. I am not sure if this is the reason display doesnt output..

sidoh, 2020-07-08:
I'm pretty sure those pin assignments are just wrong. From [Espressif's docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html), SPI bus 2 has this pinout:

```
CS   - 15
CLK  - 14
MISO - 12
MOSI - 13
```

I could be wrong, but I'm pretty sure these are not interchangeable. You could try changing them manually here --

https://github.com/sidoh/epaper_templates/blob/master/src/main.cpp#L76

let me know what you find.

xx-robot-xx, 2020-07-09:
I dont know what i am doing wrong,

I get this error while running `platformio run -e esp32 --target buildprog`

`Processing esp32 (platform: espressif32@~1.9.0; framework: arduino; board: esp32doit-devkit-v1; upload_speed: 460800; build_flags: !python3 scripts/platformio/get_version.py -DMQTT_DEBUG -Idist, -D ENABLE_GxEPD2_GFX, -D RICH_HTTP_ASYNC_WEBSERVER, -D JSON_TEMPLATE_BUFFER_SIZE=20048, -D RICH_HTTP_REQUEST_BUFFER_SIZE=20048, -D RICH_HTTP_RESPONSE_BUFFER_SIZE=20048 -D FIRMWARE_VARIANT=esp32_doit; extra_scripts: pre:scripts/platformio/build_web.py, post:scripts/platformio/build_full_image.py; lib_ldf_mode: deep+; board_build.partitions: min_spiffs.csv; lib_deps: ArduinoJson@~6.10.1, Adafruit GFX Library@~1.6.1, Timezone@~1.2.2, AsyncMqttClient@~0.8.2, ESP Async WebServer@~1.2.0, NTPClient@~3.1.0, PathVariableHandlers@~2.0.0, RichHttpServer@~2.0.0, DNSServer@~1.1.0, Bleeper@~1.0.4, xoseperez/Time#ecb2bb1, sidoh/WiFiManager#async_support, GxEPD2@~1.2.3, AsyncTCP@~1.1.1; lib_ignore: ESPAsyncTCP)`
`------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------No need to rebuild web assets.`
`CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html`
`PLATFORM: Espressif 32 1.9.0 > DOIT ESP32 DEVKIT V1`
`HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash`
`DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)`
`PACKAGES:`
 `- framework-arduinoespressif32 2.10002.190628 (1.0.2)`
` - tool-esptoolpy 1.20600.0 (2.6.0)`
`- toolchain-xtensa32 2.50200.80 (5.2.0)`
`INFO: Could not find files for the given pattern(s).
Traceback (most recent call last):
  File "scripts/platformio/get_version.py", line 32, in <module>
AttributeError: 'str' object has no attribute 'decode'
OSError: 'python3 scripts/platformio/get_version.py -DMQTT_DEBUG -Idist' exited 1:
  File "C:\Users\masked\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\masked\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\masked\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\masked\.platformio\platforms\espressif32@1.9.0\builder\main.py", line 209:
    target_elf = env.BuildProgram()
  File "C:\Users\masked\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\masked\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\masked\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\masked\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 115:
    env.ProcessFlags(env.get("BUILD_FLAGS"))
  File "C:\Users\masked\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\masked\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 213:
    env.Append(**env.ParseFlagsExtended(flags))
  File "C:\Users\masked\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\masked\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 176:
    for key, value in env.ParseFlags(str(raw)).items():
  File "C:\Users\masked\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 805:
    do_parse(arg)
  File "C:\Users\masked\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 671:
    arg = self.backtick(arg[1:])
  File "C:\Users\masked\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 589:
    raise OSError("'%s' exited %d" % (command, status))`

sidoh, 2020-07-13:
Looks like a bug in the build script -- it's choking when `git` isn't installed. Can you install `git`?

xx-robot-xx, 2020-07-15:
> Looks like a bug in the build script -- it's choking when `git` isn't installed. Can you install `git`?

Yes, I did that, and it gives me another set of errors.

[https://pastebin.com/cCRqKx4V](url)  

This is the error part (at last). ran with platformio run -e esp32 --target buildprog -v

xx-robot-xx, 2020-07-15:
Here is the full terminal output

https://pastebin.com/wHYg1qfx

sidoh, 2020-07-15:
I've run into these types of errors when the public version of the `Time` library gets installed. It stupidly includes a file named `time.h`, which obviously causes all sorts of conflicts with the standard library.

I explicitly include a patched version of `Time` to avoid this, but I think sometimes platformio pulls it in transitively.

Can you try looking in the `.pio/libdeps/esp32` directory, and deleting `Time`? There are probably several directories prefixed with `Time`, but leave those.

sidoh, 2020-07-15:
Looks from your build log (thank you for posting it, by the way, it was helpful), that may be happening:

```
|-- <Timezone> 1.2.4 (C:\Users\masked\Downloads\eink\epaper_templates\.pio\libdeps\esp32\Timezone)
|   |-- <Time> (C:\Users\masked\Downloads\eink\epaper_templates\lib\Time)
|   |   |-- <Time> 1.6 (C:\Users\masked\Downloads\eink\epaper_templates\.pio\libdeps\esp32\Time) <---------
|   |-- <Time> 1.6 (C:\Users\masked\Downloads\eink\epaper_templates\.pio\libdeps\esp32\Time)
```

sidoh, 2020-07-15:
(by the way -- this should obviously not be happening at all. hopefully a way to convince platformio to never include `Time`. just meaning to unblock you)

xx-robot-xx, 2020-07-16:
> I've run into these types of errors when the public version of the `Time` library gets installed. It stupidly includes a file named `time.h`, which obviously causes all sorts of conflicts with the [...]

Thank you so much :)  that solved the problem! I was able to compile.

But after flashing i still have the same problem, no display output, i changed

[https://github.com/sidoh/epaper_templates/blob/master/src/main.cpp#L76](url)

to values mentioned in this link

[https://github.com/ZinggJM/GxEPD2/blob/master/examples/GxEPD2_WS_ESP32_Driver/GxEPD2_WS_ESP32_Driver.ino#L97](url)

I tried examples from the GxEPD2 library, the waveshare demo and accent.ink ([https://github.com/maxbbraun/accent](url)) and i am able to run those on the waveshare esp32 driver board.

sidoh, 2020-07-16:
Excellent -- glad that got the compile working.

Note that as-is, that section in `main.cpp` only applies if the SPI bus setting is set to `VSPI` (which is not the default). So you'll either want to remove the conditional, or change the setting in the web UI.

gardenali, 2020-12-20:
So, what's the state on this bug? I just bought the waveshare driver board and ran into this issue.

sidoh, 2020-12-21:
Hi, @gardenali -- I just added a setting that should allow you to select the SPI pin configuration used by the waveshare board. More info in this comment:

https://github.com/sidoh/epaper_templates/issues/53#issuecomment-748698704

If you could try out the beta release and let me know if it works, that'd be helpful.

gardenali, 2020-12-21:
It doesn't seem very happy to load that:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:588500
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:588500

Flashing went fine as usual. Flashed to 0x1000 following the documentation. And of course I did gunzip the file.

Connecting........___
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: XXXXXXXXXXXX
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 1716352 bytes to 1034197...
Wrote 1716352 bytes (1034197 compressed) at 0x00001000 in 27.7 seconds (effective 496.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

sidoh, 2020-12-21:
Try flashing with the [INITIALIZER release from the latest version](https://github.com/sidoh/epaper_templates/releases/download/v2.5.1/INITIALIZER_epaper_templates_esp32-v2.5.1.bin)

Then use the web UI to upload the beta binary. The INITIALIZER release is built by travis and includes some extra stuff that the usual releases don't.

gardenali, 2020-12-21:
You beat me to it. I finished work, flashed the INITIALIZER, updated it, and saw your comment right after that. Thanks :)

No luck yet, I'm afraid. Let me know what I can do to help and I'll try to diagnose further.

The switch is set to position B, which is according to their manual for my display type (Waveshare 7.5" v2 B/W, 800x480). This works with their demos:
[image]

I'm guessing the right model to choose here is "GDEW075T7 (7.5" B/W 800X480)" but I'm not entirely sure. This could be the issue. However, I don't see another display option that's compatible, and I don't have another model to test.

I defined a very simple template, to avoid any issues. I defined a formatter as it was complaining with "WARNING: formatter definition block either missing or is of invalid type." before I created one. Still unsure how that's used, but I defined it.
{
  "lines": [
    {
      "x1": 69,
      "y1": 62,
      "x2": 572,
      "y2": 367,
      "color": "black"
    }
  ],
  "formatters": [
    {
      "formatter": {
        "args": {},
        "type": "identity"
      },
      "name": "asd"
    }
  ],
  "background_color": "white",
  "rotation": 0,
  "bitmaps": []
}

When I tried Busy 25, DC 27, RST 26 (suggested above), I got this output:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: xxxxxxxxxxx
Applying settings
*WM: freeing allocated params!
Loading new template: /t/Test1.json
Loading template: /t/Test1.json
Free heap - 140600
formatter key = asd
_PowerOn : 6
_Update_Full : 1
_PowerOff : 1

When I tried Busy 10, DC 12, RST 11 (suggested on the other bug by dissy614), I got this output:
"GPIO 25, pin 10, Busy
GPIO 26, pin 11, RST
GPIO 27, pin 12, D/C
(Here "Pin" is on the ESP-32s chip)"

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: xxxxxxxxxxxxxx
Applying settings
*WM: freeing allocated params!
Loading new template: /t/Test1.json
Loading template: /t/Test1.json
Free heap - 140436
formatter key = asd
_PowerOn : 7
_Update_Full : 1
_PowerOff : 1

When I tried Busy 19, DC 22, RST 21 which I tried to guess from this snippet in their bluetooth demo that works, just to see if I had something there:
#define PIN_SPI_SCK  13
#define PIN_SPI_DIN  14
#define PIN_SPI_CS   15
#define PIN_SPI_BUSY 25//19
#define PIN_SPI_RST  26//21
#define PIN_SPI_DC   27//22

What I got on the serial log:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: xxxxxxxxxx
Applying settings
*WM: freeing allocated params!
Loading new template: /t/Test1.json
Loading template: /t/Test1.json
Free heap - 140584
formatter key = asd
Busy Timeout!
_PowerOn : 10001058
Busy Timeout!
_Update_Full : 10001017
Busy Timeout!
_PowerOff : 10001014

sidoh, 2020-12-21:
I think that's the right model. I have that display and have gotten it to work. I don't have the ESP32 driver you're using (I did order one, not sure when it will be here.)

How are you updating the pins? Just defining those constants is probably not enough, but that may not be what you're trying. In case it's useful, here's a spot to fiddle with SPI pins:

https://github.com/sidoh/epaper_templates/blob/master/src/main.cpp#L77

(will have to have the bus set to VSPI for this to work, of course)

gardenali, 2020-12-21:
I was just changing these pins on the UI, not defining any constants. I was hoping that the UI change would be enough.
I'll go read the source some other time.

sidoh, 2020-12-21:
Ah ok -- yeah, rst, busy, and dc are all configurable via the UI. The issue here is the SPI bus, though. You've selected the "Waveshare" option there right?

gardenali, 2020-12-21:
Yep. I did.

sidoh, 2020-12-21:
Cool, thanks for trying it! It's a little difficult to see what space there is left between the example sketch and the epaper_templates code (but obviously there's something there given the example is working).

Might have to wait until I have the hardware to replicate. Obviously happy to look at a PR as well!

gardenali, 2020-12-22:
Found it. Sorry I won't have the time to cook an elegant patch, but here's the idea:
1. From the example at lines 94-99 from
https://github.com/ZinggJM/GxEPD2/blob/49224e62fd7837bbe2800ba8c4d2c64a5886e562/examples/GxEPD2_WS_ESP32_Driver/GxEPD2_WS_ESP32_Driver.ino#L94
   ```SPI.end()``` and ```SPI.begin(13,-1,14,15)``` or the given example ```(13,12,14,15)``` which also works *MUST* happen AFTER ```display->init(115200);```  https://github.com/sidoh/epaper_templates/blob/9c94a7d6f0837ac3350ae819ce460e0c37c1b0ec/lib/Display/DisplayTemplateDriver.cpp#L27
2.   __gxepd2_build_bw_driver() assumes SS. I had to replace SS with 15 here:  https://github.com/sidoh/epaper_templates/blob/9c94a7d6f0837ac3350ae819ce460e0c37c1b0ec/lib/Display/DisplayTypeHelpers.cpp#L11
  ```return new GxEPD2_BW<Display, Display::HEIGHT>(Display(15, dc, rst, busy));```

[image]

sidoh, 2020-12-22:
Super helpful, thanks for digging at it! I'll try to get this change in soon.

sidoh, 2020-12-23:
Made the following changes as you suggested:

* Initialize SPI _after_ `display.init`
* Use correct SS/CS pin for SPI bus selection (15 for Waveshare)

Here's a pre-compiled release: [epaper_templates-waveshare_driver_spi_setting.bin.gz](https://github.com/sidoh/epaper_templates/files/5736905/epaper_templates-waveshare_driver_spi_setting.bin.gz)

Changes are being pushed to [this branch]https://github.com/sidoh/epaper_templates/tree/feature/waveshare_driver_spi_setting) if you'd prefer to build locally.

dissy614, 2020-12-24:
@sidoh unfortunately still nothing with 2.5.1-12-g7d9aad0 for me.
Strange too, as my display is the model prior to gardenali's (7.5" b&w, but at 640x384 instead of the higher 800x res)

sidoh, 2020-12-24:
@gardenali, let me know if the release is working.

Here is the relevant commit -- if you notice anything that's off direction from your local changes that do work, let me know.

gardenali, 2020-12-24:
Neither 2.5.1-12-g7d9aad0 nor v2.6.0-rc.1 worked for me initially, or so I thought. After leaving it sitting around for a while with 2.5.1-12-g7d9aad0, it DID update the screen (5-10 mins sitting idle, no visible refresh)

I then went to the log, and it's crashing left and right:
```
_PowerOn : 9
_Update_Full : 1
_PowerOff : 1
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x°ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
cets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:110üets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.239.125
Applying settings
MqttClient - connecting
*WM: freeing allocated params!
MqttClient - subscribing to topic: template-displays/display1/+
Loading new template: /t/aaa.json
Loading template: /t/aaa.json
Free heap - 138272
Rendering bitmap: /b/wifi, x=731, y=5, w=64, h=64
_PowerOn : 9
_Update_Full : 1
_PowerOff : 1
*r…ÆP±µ±…,µ«m½¢À-X±[02]¡^&å±§F[19]dpéù®±R[0B]èbådbv#‰%¹á¹f8á~ÑèrÁ-±@%VF-±dN8|út«X±±È'p×,P©wªt‹,±È-×¯±P©Ñ'p×,P©Ç'p×,P‰X~§-jqè|É)©EX[!ù(X±+tÇl±ðªan êÉª±R·+±"\Ù«±[02]±êÉª±T±Å,±ðªa¬µ[!ù(X±Å,±Å,ðªa/±[a|×,±T±±…l *WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.239.125
Applying settings
[... 127 lines cut ...]
_PowerOn : 9
_Update_Full : 1
_PowerOff : 1
*r…ÆP±µ±…,µ«m½¢À-X±
ènNfj$f†ª~Ñ-±"¡ˆd¹cbv#‰%¹á¹f8á~ÑèrÁ-±@%VF-±dN8|út«X±±È'p×,P©wªt‹,±È-×¯±P©Ñ'p×,P©Ç'p×,P‰X~§-jqè|É)©EX[!ù(X±+tÇl±ðªan êÉª±R·+±"\Ù«±[02]±êÉª±T±Å,±ðªa¬µ[!ù(X±Å,±Å,ðªa/±[a|×,±T±±…l *WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.239.125
Applying settings
MqttClient - connecting
*WM: freeing allocated params!
Loading new template: /t/aaa.json
Loading template: /t/aaa.json
Free heap - 138584
Rendering bitmap: /b/wifi, x=731, y=5, w=64, h=64
_PowerOn : 125001
MqttClient - subscribing to topic: template-displays/display1/+
_Update_Full : 4074004
_PowerOff : 40001
```

The template has a simple 64x64 bitmap, a single line and the string "asd".

gardenali, 2020-12-24:
My crude hack that worked (don't try this at home, kids!)

```
--- a/lib/Display/DisplayTemplateDriver.cpp
+++ b/lib/Display/DisplayTemplateDriver.cpp
@@ -25,6 +25,9 @@ DisplayTemplateDriver::DisplayTemplateDriver(
 
 void DisplayTemplateDriver::init() {
   display->init(115200);
+  Serial.println(F("HACKY HACK2"));
+  SPI.end();
+  SPI.begin(13, 12, 14, 15);
   display->mirror(false);
   vars.load();
 }
```

```
--- a/lib/Display/DisplayTypeHelpers.cpp
+++ b/lib/Display/DisplayTypeHelpers.cpp
@@ -9,12 +9,12 @@
 
 template <class Display>
 inline static GxEPD2_GFX* __gxepd2_build_bw_driver(const uint8_t dc, const uint8_t rst, const uint8_t busy) {
-  return new GxEPD2_BW<Display, Display::HEIGHT>(Display(SS, dc, rst, busy));
+  return new GxEPD2_BW<Display, Display::HEIGHT>(Display(15, dc, rst, busy));
 }
 
 template <class Display>
 inline static GxEPD2_GFX* __gxepd2_build_3c_driver(const uint8_t dc, const uint8_t rst, const uint8_t busy) {
-  return new GxEPD2_3C<Display, Display::HEIGHT>(Display(SS, dc, rst, busy));
+  return new GxEPD2_3C<Display, Display::HEIGHT>(Display(15, dc, rst, busy));
 }
```

And then I could (or not, irrelevant) take out the SPI.end() and SPI.begin() from main.cpp.

@dissy614 In case you want to play with binaries offered by random strangers on the internet, feel free to try this one out.
NOTICE: HARDCODES MAKE IT NOT WORK WITH OTHER BOARDS, IT IS NOT AN ELEGANT PIECE OF CODE AND I OFFER NO WARRANTY. IF ANYTHING BREAKS, IT'S YOUR FAULT :)

[firmware.bin.gz](https://github.com/sidoh/epaper_templates/files/5740327/firmware.bin.gz)

dissy614, 2020-12-24:
@gardenali Thank you, and yes, that fw booted up right away drawing my template.  Working with the GDEW075T8.

sidoh, 2020-12-24:
With the 2.6.0-rc.1 release, you guys selected the Waveshare SPI bus right?

dissy614, 2020-12-24:
> With the 2.6.0-rc.1 release, you guys selected the Waveshare SPI bus right?

I did, Waveshare and 25=busy,26=rst,27=dc
Also tried changing it to default, reboot, changing it back, reboot.   Upon loading @gardenali 's binary, those same settings were retained and worked instantly after boot.

I'll give it a full SPI flash erase and start with the initializer image

dissy614, 2020-12-24:
@sidoh Something is still amiss in the rc1.
Fully erased the flash and started over with the rc1 initializer.  Set everything up from my notes and made a testing template.  No reaction from the screen.
In the WebUI, put back gardenali's 2.5.1-11 based fw, reboot, and the screen updates with the saved settings.  Put on the rc1 fw, set to a blank template, and nothing updated.  Verified settings were still retained.

Quite strange

sidoh, 2020-12-24:
I'm sure I'm missing something in it, then. Haha.

I'm trying to connect my spare screen & ESP32 using the Waveshare pinout. Hopefully can figure this out soon.

dissy614, 2020-12-24:
@sidoh I too am flabbergasted how and why Waveshare would do this.
I'm about > < close to breaking out the hotair rework and bodge-gauge wire to "fix" their connector, while waiting on amazon to deliver a non-freak esp32 ;}

sidoh, 2020-12-24:
I'm a dolt. I went through all the effort of making the SS pin pluggable and forgot to pass it into the GxEPD2 initialization call. (fixed in https://github.com/sidoh/epaper_templates/commit/059daa47e1e53a556322ab7c0285f3a6b152798c).

2.6.0-rc.2 should be working. Actually tested it this time:

* Set up my spare screen & ESP32 with the Waveshare pins
* 2.6.0-rc.1 was not working for me
* @gardenali's release was
* 2.6.0-rc.2 does work

Travis usually takes 10-20 minutes to start building these days, but pre-compiled release should be up after that.

dissy614, 2020-12-24:
@sidoh woo!  Confirmed, rc2 works

sidoh, 2020-12-24:
Great! I'm also going to add a setting to directly override the SS pin. My setup uses HSPI with SS = 5.

That's probably everything for this release, so if it's working for you guys I'll cut the non-RC version sometime later today.

dissy614, 2020-12-24:
Awesome news.  Thank you so very much for all your efforts and time on this!  
I'll let the thread end by saying I absolutely love your template system too.  All the best!

gardenali, 2020-12-24:
Just tried 2.6.0-rc3 with the new override SS pin input. Works!
Not sure what's up with ntp though, but that's a different bug.  I'll review that and see if it needs anything later.

```
Formatted value: 13:39
Rendering timestamp
_PowerOn : 125003
_Update_Part : 1572001
_PowerOff : 40001
Formatted value: 13:38
Rendering timestamp
_PowerOn : 125004
_Update_Part : 1572001
_PowerOff : 40001
Formatted value: 13:39
Rendering timestamp
_PowerOn : 125001
_Update_Part : 1572001
_PowerOff : 40001
```

This seems fixed. Thanks a lot for your help!

And yes, I second the mention that this template system is quite nice!

sidoh, 2020-12-25:
Fixed in 2.6.0.

closed 2020-12-25 by sidoh