sidoh/epaper_templates #47: Variables list empty after refreshing/leaving page

skitterrusty, 2020-10-31:
After adding a variable and saving it, then navigating away or refreshing the page, the variables list will be empty upon return.
The saved variables can still be found if the variable name is used in a template.
NodeMCU32-S

skitterrusty, 2020-10-31:
Fixed after reflash - weird

skitterrusty, 2020-11-02:
Re-occurred. Not sure what triggered it. Variables still function properly, just don't show on the variables page.

skitterrusty, 2020-11-02:
Appears to be issue with pagination logic. Browser network log shows that the script creates an infinite number of pages of variables. Causes additional power usage by ESP chip and can eventually crash it.

sidoh, 2020-11-02:
Thanks for opening -- I'm seeing this too. Will try to look when I have more time.

appi1, 2020-11-13:
Hello great app thanks
even in my installation the variables are not visible in the Variables tab. Unfortunately I cannot send any values via Mqtt to e.g. Send the variable Outside_temperature, although the value arrives as subscriber in MQTT.fx. the subscriber function doesn't seem to work with version v2.5.1

sidoh, 2020-11-13:
Hi @appi1, I'm guessing that's a separate problem. Please open another issue with the output of `http://<you_epaper_ip_address>/api/v1/settings` (**make sure to sensor any passwords that appear here**).

sidoh, 2020-12-24:
Hi @skitterrusty, an update on this --

I think the issue is some kind of corruption of the variables database file. I haven't been able to reproduce it, unfortunately, but it made my display mostly unusable until it was resolved.

The easiest fix for me was to clear the database. The `DELETE /api/v1/variables` route does this (example command: `curl -X DELETE http://epaper-display/api/v1/variables`), but it wasn't aggressive enough. The [2.6.0-rc.1 release](https://github.com/sidoh/epaper_templates/releases/tag/v2.6.0-rc.1) has a more aggressive delete which worked for me.

Obviously not ideal, and definitely need to fix the underlying issue, but hopefully this offers a temporary workaround.

My guess is the issue is some kind of race condition.