sidoh/epaper_templates #48: variables cannot be updated with MQTT

appi1, 2020-11-14:
hi
I send a value with MQTT-fx to template-displays / display1 / outside_temp but the variable is not updated.
on my mobile device with IoT panel, the value of the variable is correctly adjusted. When I adjust the variable in the UI, the update takes place on the display

txh 

my configuration is:
{"display.display_type":"GDEW042T2","display.full_refresh_period":"3600000","display.template_name":"/t/my_display.json","display.windowed_updates":"false","hardware.busy_pin":"7","hardware.dc_pin":"17","hardware.rst_pin":"16","hardware.spi_bus":"HSPI","mqtt.client_status_topic":"template-displays/my_display/status","mqtt.password":"","mqtt.server":"mqtt-broker-rut955","mqtt.username":"","mqtt.variables_topic_pattern":"template-displays/my_display/:outside_temp template-displays/my_display/:inside_temp","network.hostname":"epaper-display","network.mdns_name":"epaper-display","network.setup_ap_password":"waveshare","network.wifi_password":"","network.wifi_ssid":"","power.awake_duration":"30","power.sleep_duration":"600","power.sleep_mode":"ALWAYS_ON","power.sleep_override_pin":"25","power.sleep_override_value":"1","system.timezone":"CET","web.admin_password":"","web.admin_username":"","web.port":"80"}

sidoh, 2020-11-15:
Looks like your setting for `mqtt.variables_topic_pattern` isn't quite right.

It should be `template-displays/bart/:variable_name`. The `:variable_name` is a _placeholder_ that causes the client to subscribe to `template-displays/bart/*`.

appi1, 2020-11-16:
sorry, it was mistake in my definition.......
thx

closed 2020-11-16 by sidoh

creathinq, 2022-06-06:
Hi. I can't get MQTT to update variable, there is something I'm missing... I can't really work out the exact relations between variable, mqtt.variables_topic_pattern and node-red topic

my config:
`{"display.display_type":"GDEH0213B72","display.full_refresh_period":"10000","display.template_name":"/t/a.json","display.windowed_updates":"false","hardware.busy_pin":"4","hardware.dc_pin":"17","hardware.rst_pin":"16","hardware.spi_bus":"HSPI","hardware.ss_pin_override":"5","mqtt.client_status_topic":"","mqtt.password":"","mqtt.server":"192.168.0.179:1883","mqtt.username":"","mqtt.variables_topic_pattern":"template-displays/epd/:temp1","network.hostname":"epd","network.mdns_name":"epd","network.ntp_server":"pool.ntp.org","network.setup_ap_password":"waveshare","network.wifi_password":"******","network.wifi_ssid":"******","power.awake_duration":"30","power.sleep_duration":"600","power.sleep_mode":"ALWAYS_ON","power.sleep_override_pin":"25","power.sleep_override_value":"1","system.timezone":"CET","web.admin_password":"","web.admin_username":"","web.port":"80"}`

So there I have `"mqtt.variables_topic_pattern":"template-displays/epd/:temp1"` and node-red is publishing msg to topic `template-displays/epd/t1`

there is funtion block in node-red that extracts temperature from ds18b20 temp sensor:` return {payload: msg.payload ["DS18B20"]. Temperature};`

debug node says msg is
`
 :msg.payload : number
26.4
`

and I have variable named "temp1", but it is empty no matter what I try. mqtt_state says connected.

I also have this from serial console:
```
MqttClient - connecting
MqttClient - subscribing to topic: template-displays/epd/:temp1
Loading new template: /t/a.json
Loading template: /t/a.json
Free heap - 170428
formatter key = formattter
Rendering temp1
_PowerOn : 91001
_Update_Full : 1690001
_PowerOff : 173001

```
I`m on version 2.6.0. pls help :)