sidoh/epaper_templates #52: M5Paper support?

henriklied, 2020-12-04:
This looks like a really cool project! I'm planning on testing it with my Waveshare 7.5" later today, but I was wondering what your thoughts are about getting this working with the newly released [M5Paper](https://m5stack.com/products/m5paper-esp32-development-kit-960x540-4-7-eink-display-235-ppi?variant=37595977908396)? It looks like a cool piece of kit with a great screen.

sidoh, 2020-12-04:
Hey @henriklied, thanks for your interest!

Looks like a really cool product! Would be especially cool to incorporate support for the touch events.

I'm using the [GxEPD2](https://github.com/ZinggJM/GxEPD2) library to handle the SPI comms with the display. It only supports Waveshare/Good Display boards, and I expect that will continue to be the case.

Happy to collaborate on support for other boards, but it'd be a pretty large effort to support a non-Waveshare board.

sidoh, 2020-12-24:
Going to close this since it seems out of the scope I see this project having for the foreseeable future, but happy to collaborate on support for non-Waveshare boards!

closed 2020-12-24 by sidoh