sidoh/epaper_templates #53: Unable to connect my waveshare epaper-hat / not saving changes to eeprom at all

arnoldu, 2020-12-05:
I have spent pretty three hours to get epaper-display working with my setup:

Adafruit Feather / Waveshare 800x480xB/W/R with Waveshare epaper-hat.
epaper-display V2.5.1
This HW configuration is working with other epaper-libraries quite well.

I can't get it to work :

Whenever I change something on the hardware settings page and press save the serial debug stream says "Applying settings". When I reboot then, either over the webinterface or pressing the reset button the values have not been saved! This is even true with other settings, e.g. timezone.

I even tried another ESP32 board (Joyit NODEMCU clone) with the same results.

arnoldu, 2020-12-05:
Another observation:

After changing anything , e.g. timezone and pressing save, the serial monitor shows "Applying settings" and the save buttons in the web interface stay grayed!

arnoldu, 2020-12-05:
I have made a short screencast and attach it here
[image]

arnoldu, 2020-12-05:
as you can see the error is the same with version 2.5.0

sidoh, 2020-12-05:
The save button being greyed out for that long probably means the settings aren't being applied at all. Only reason I can imagine that'd happen is if the chip were initialized with different SPIFFS parameters than the firmware is built for.

Can you share the command/procedure you used to flash the chip?

arnoldu, 2020-12-05:
I use my good ol' esptool.py:

`esptool.py --chip esp32 --baud 460800 write_flash 0x1000 INITIALIZER_epaper_templates_esp32-v2.5.0.bin`

arnoldu, 2020-12-05:
I investigated further:
I took another brand new ESP-board (https://www.waveshare.com/w/upload/8/80/E-Paper_ESP32_Driver_Board_Schematic.pdf) It features an external connector the Waveshare Display 800x480 RBW.
The firmware 2.5.0 behaves a little different here - all settings can be saved, the save button gets ungrayed after a second EXCEPT the Hardware settings they are simply not saving. After a reboot the old default values are still there.
Strangely you have in your README GPIOS 12,14,15 for HSPI, the schematic (see enclosed schematic) shows clearly pins 13,14,15. I am confused here ...

sidoh, 2020-12-06:
Gotcha -- command looks good to me.

It's strange that you can save some settings and not others. I don't know why that'd be happening. Any difference in the serial output? 

The flow for saving settings via the web interface is:

* Merge the settings patch into existing settings in memory
* Dump settings object to disk
* Live apply settings

Can you maybe paste the curl command from your browser's dev tools corresponding to the failing request?

For what it's worth -- I'm able to save hardware settings. Actually managed to lock myself out of my install because having the wrong busy pin causes a busy loop on startup :X.

12 is MISO and 13 is MOSI. If I'm reading the schematic you linked correctly, it has GPIO 14 labeled as SDI (MOSI), but GPIO 14 is the HSPI CLK. So not sure what's going on there either, honestly.

arnoldu, 2020-12-06:
I experimented with demo code to find out the right SPI pins:

The following assignments are working, but are different from your pinouts:

#define CS_PIN           15
#define CLK_PIN          13
#define DIN_PIN          14

But why the heck?
I will try to compile your code with these pins compiled in and let you know what happened

dissy614, 2020-12-07:
Would there be any possibility of a new binary release , either with an option to use the waveshare esp32 driver (as defined above), or the latest GxEPD2 which recently added support to SPIClass remap pins for this board?

sidoh, 2020-12-07:
Yeah, would definitely want to make the SPI pins configurable if they're not fixed. Seems like what's happening is either:

1. I'm misreading the ESP32 datasheet & other online resources, and the HSPI pins I have in my code are wrong
2. The HSPI pins are not always the same

I want to get this fixed for you guys, but obviously the changes for (1) and (2) are different. 

My install actually uses HSPI with the pin mappings I have configured and has been mounted to my wall & working for more than a year, so it seems like maybe (2) is the case. I would not expect the SPI CLK to be a variable pin, but maybe that's not true.

If you guys could confirm things are working with the pins switched around as shown in the schematic it'd be very helpful.

sidoh, 2020-12-07:
(@arnoldu -- the build should be working again)

dissy614, 2020-12-08:
@sidoh Unfortunately I only have the Arduino IDE at my disposal.  and on windows no less :{
Have had a h* of a time with the dependencies for platform.io, so not sure if I can be helpful or not.

If I use your VSPI settings, and wire an adaptor to the waveshare board headers, it works.
If I use your HSPI settings, and wire an adaptor to the waveshare board headers, it does not work, 
(because I do not have access to IO 15? it is tied low it seems)

I am using your v2.5.1 binary release (init_ version) with espressif "flash download tools"

The waveshare example code does not use the SPI library, they bang out bytes to the display.
In the Arduino IDE they use:
  #define PIN_SPI_SCK  13
  #define PIN_SPI_DIN  14
  #define PIN_SPI_CS   15
  #define PIN_SPI_BUSY 25
  #define PIN_SPI_RST  26
  #define PIN_SPI_DC   27

Also for reference, the on-board connector is wired as:
  GPIO 13, pin 16, SCLK
  GPIO 14, pin 13, SDI (or DIN)
  GPIO 15, pin 23, CS
  GPIO 25, pin 10, Busy
  GPIO 26, pin 11, RST
  GPIO 27, pin 12, D/C
(Here "Pin" is on the ESP-32s chip)

GxEPD2 v1.2.13 added remapping explicitly for this board in v1.2.13  (Oct 5th)
// NOTE: this board uses "unusual" SPI pins and requires re-mapping of HW SPI to these pins in SPIClass
//       this example shows how this can be done easily
https://github.com/ZinggJM/GxEPD2/blob/master/examples/GxEPD2_WS_ESP32_Driver/GxEPD2_WS_ESP32_Driver.ino

I would be happy to test pre-compiled images

sidoh, 2020-12-08:
Super helpful background, @dissy614. tyvm.

I'll check out the GxEPD2 updates. Do you guys have a link to the board you're using?

dissy614, 2020-12-08:
@sidoh 
Product: https://www.waveshare.com/e-paper-esp32-driver-board.htm
Wiki: https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board
The later has the schematic (the PDF that arnoldu linked above), and their Arduino sample code (that doesn't use SPIClass)

They also link to a (silly huge) fourm thread by ZinggJM who wrote GxEPD.  It was somewhere around page 140-145 or so, that ended up with the .ino link above (which was far more useful to me)

sidoh, 2020-12-08:
Very helpful, thanks.

Yeah, just dropped into that forum thread to report a bug yesterday actually. :D

dissy614, 2020-12-08:
Oh, and the adaptor I mentioned, is actually this: https://www.waveshare.com/wiki/E-Paper_Driver_HAT
Technically an e-Ink to rasp-pi board, but with an 8 pin SPI breakout on the side, which is what I wired to the ESP32 boards header pins for testing

sidoh, 2020-12-21:
Hi @dissy614, I added a setting to configure the SPI pins to be the ones used in the example you linked.

Can you try out the attached build and let me know if it works? 

[image]

[epaper_templates-waveshare_driver_spi_setting.bin.gz](https://github.com/sidoh/epaper_templates/files/5721769/epaper_templates-waveshare_driver_spi_setting.bin.gz)

dissy614, 2020-12-23:
> Hi @dissy614, I added a setting to configure the SPI pins to be the ones used in the example you linked.
> Can you try out the attached build and let me know if it works?

Unfortunately it doesn't seem to be working.  I've been a bit short on free time, but the next few days coming up I can do a bit more in-depth debugging

sidoh, 2020-12-23:
Hi @dissy614, iterating with help from @gardenali. There's a new release up at the bottom of https://github.com/sidoh/epaper_templates/issues/41. Let me know if it works.

Ordered the Waveshare ESP32 driver, but hasn't shipped yet.

sidoh, 2020-12-25:
Should be fixed in v2.6.0.

closed 2020-12-25 by sidoh