sidoh/epaper_templates #54: Code wont compile without errors

arnoldu, 2020-12-06:
I try to compile the latest version on my Ubuntu 20.04 with Python3. I' d expect to compile without errors what I received instead was:

`git clone https://github.com/sidoh/epaper_templates.git`
`platformio run -e esp32 --target buildprog`

to reveive error in main.cpp:
src/main.cpp:30:25: fatal error: WiFiManager.h: No such file or directorysrc/main.cpp:30:25: fatal error: WiFiManager.h: No such file or directory`

when I try to install the WiFiManager with
`platformio lib install WiFiManager` 

I receive lots of missing 8266 libs missing. I guess WiFiManager is a lib for 8266 only.

How to proceed?
Do you need any other information from me?

sidoh, 2020-12-06:
Hey @arnoldu -- this is due to some changes in how PlatformIO manages dependencies. Ran into this issue in other projects I maintain.

I'll push a fix shortly.

sidoh, 2020-12-06:
Pushing a fix for the libraries, but there's a compile error in an upstream library (GxEPD2). I pinned the minor version, but a patch version introduced the error. Will pin the patch version for now as well.

sidoh, 2020-12-06:
Also posted on GxEPD2's forum thread -- https://forum.arduino.cc/index.php?topic=487007.msg4817679#msg4817679

sidoh, 2020-12-06:
hopefully fixed, lmk if you're still having trouble

closed 2020-12-06 by sidoh