sidoh/epaper_templates #57: esp32 pins ?

digitalurban, 2020-12-22:
Afternoon - I am failing to connect my 7.5 b/w waveshare - i can't for the life of me figure out the pins - its a node esp32 board with this mapping:

https://cdn.shopify.com/s/files/1/1509/1638/files/ESP_-_32_NodeMCU_Developmentboard_Pinout_Diagram.jpg?4479111012146266271

On the examples, there is not a note on where the DIN goes or the CS? Also do Pins map to GPIO? 

This may be a stupid question but any help would be great, i have the interface running, just no connection to the screen.

Andy

sidoh, 2020-12-22:
Hi @digitalurban,

The pinouts I use can be found here: https://github.com/loboris/ESP32_ePaper_example#connecting-the-display

"DIN" is MOSI, I believe. Yes, the pin numbers are the same as the GPIO numbers in the schematic you linked.

The documentation for this project could be more helpful in this regard. :)

sidoh, 2020-12-25:
Hey @digitalurban -- improved the hardware setup section in the README:

https://github.com/sidoh/epaper_templates#hardware-setup

Let me know if you have questions!

closed 2020-12-25 by sidoh