sidoh/epaper_templates #67: Support for LOLIN TFT / Driver IC: IL3897 ??

bwente, 2021-01-18:
I trying to connect a LOLIN 2.13 inch display to a LOLIN D32 Pro. I am using the TFT cable to connect them, so it is pretty much plug and play.

I can't seem to get the right combination of setting to get anything to display. Is the e-paper supported?

sidoh, 2021-01-19:
I'm not familiar with the display board. If you know what the pins are _supposed_ to map to, I'm happy to help out or take a loot.

bwente, 2021-01-20:
The display screen is this one...
2.13 inch e-paper display module partial refresh e-ink screen, GDEH0213B73 (EOL)
https://www.good-display.com/product/362.html

I tried the pins suggested here.
https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/issues/19#issuecomment-450844686

and here
https://forum.micropython.org/viewtopic.php?f=18&t=7547&sid=5c3aa022424b6561dd65f50409dd74dc

bwente, 2021-01-20:
The ESP32 board...
[image]

bwente, 2021-01-22:
Based on the micropython.org post, it seems the TFT cable is mapped like this.

MOSI = PIN(23)
SCK = PIN(18)
CS = PIN(14)

DC = PIN(27)
RST = PIN(33)
BUSY = PIN(32)

MISO = PIN(19)

0raculo, 2022-08-07:
Anyone got this to work?

0raculo, 2022-08-10:
Can confirm this works nicely!