sidoh/epaper_templates #69: [Feature request] Can you add GDEH0154D67 (Supported by GxEPD2)

velaar, 2021-02-03:
I tried to add it directly to lib/Display/DisplayTypeHelpers.cpp but doesn't really work. Maybe I'm missing something.

sidoh, 2021-02-04:
Hey @velaar, happy to help add support for a new GxEPD2 display type.

Could you link a commit or a diff so that I know what you tried, and provide a little more information about the way in which it didn't work?

velaar, 2021-02-04:
Hey @sidoh 
Here is what I tried: 
[diff.txt](https://github.com/sidoh/epaper_templates/files/5928880/diff.txt)

It seems the display doesn't initialize at all. Using HSPI on ESP32 (ESP32-PICO-KIT). The display works fine with EPD2 examples.

sidoh, 2021-02-05:
Those changes look correct to me. Anything interesting in Serial after you select that display type?

(disclaimer: it's going to be a little difficult for me to give detailed help because I don't own this module.)