sidoh/epaper_templates #70: full customisation of pins

matphillips, 2021-02-17:
I have a couple of epaper displays that are housed in a 3d printed enclosure, but the enclosure is glued shut. the MOSI pin is connected to ESP pin 23.
As such, the display cant be used with this software because it wants MOSI on 13,14 or 19. 

Is there a way to customise the pin mappings?

sidoh, 2021-02-18:
Hey Matt,

I was originally under the impression that the SPI pins were not configurable on the ESP32, but it's sounding like that's not true.

Presently not a way to customize SPI pins beyond selecting one of the busses (HSPI/VSPI), but would be happy to add a way or collaborate on a PR.

If you have working examples of GxEPD2 sketches using the non-standard pins that'd be a big help.

matphillips, 2021-02-18:
Works with the standard GxEPD2_32 library using Arduino IDE, if I remember right, the pins are defined by the board definition files, and do seem to change a bit between them.
Also works with ESPHome with the platform set to 'waveshare_epaper' and defining the spi pins in the yaml file to whatever is needed.

Pin 23 is the MOSI pin for quite a few esp32 boards it seems.