sidoh/epaper_templates #76: [Unexpected Webpage Error] Cannot read property '0' of undefined

Ianszh, 2021-04-20:
```json
{"error":{"message":"Cannot read property '0' of undefined","stack":"TypeError: Cannot read property '0' of undefined\n    at J (http://192.168.1.195/assets/index.6fe41c4e.js:1:9787)\n    at J (http://192.168.1.195/assets/index.6fe41c4e.js:1:9784)\n    at http://192.168.1.195/assets/index.6fe41c4e.js:1:47009\n    at Array.forEach (<anonymous>)\n    at http://192.168.1.195/assets/index.6fe41c4e.js:1:46998\n    at e.produce (http://192.168.1.195/assets/2.41e49d68.js:20:1367)\n    at http://192.168.1.195/assets/index.6fe41c4e.js:1:46973\n    at http://192.168.1.195/assets/index.6fe41c4e.js:1:41138\n    at Object.onChange (http://192.168.1.195/assets/index.6fe41c4e.js:1:38408)\n    at t.value (http://192.168.1.195/assets/2.41e49d68.js:57:5613)","name":"TypeError"}}
```

Ianszh, 2021-04-20:
Hello Chris, I managed to get the ESP flashed, while trying with 7.5 inch on the 12.48 inch panel I ran into this error.

I'm not sure if it's relevant

sidoh, 2021-04-20:
Was this after formatting and re-flashing the ESP? Can you share how you formatted the flash?

Some additional output that might help:

1. /api/v1/settings (sensor passwords!)
2. /api/v1/templates
3. /api/v1/system