sidoh/epaper_templates #78: use different Locale (IT)

tigroz, 2021-05-04:
I am very impressed with the work you have done, I wanted to know if it is possible to use another language to display the names of the days of the week and the months (Italian) I have already changed some settings to display the temperature in Celsius, could you help me?
[image]

sidoh, 2021-05-04:
Hi @tigroz!

This is a good question. I'm using the native [strftime](https://github.com/sidoh/epaper_templates/blob/master/lib/Variables/TimeFormatter.cpp#L38) for this functionality. At least in vanilla C, stuff like day of week is [locale-specific](https://www.cplusplus.com/reference/ctime/strftime/), which is configured through [`setlocale`](https://www.cplusplus.com/reference/clocale/setlocale/).

I have no idea if the ESP32 SDK ships with locales other than English. I would hope it does, but I'm not seeing a ton of information about it.

I'd be happy to experiment with this myself, but if you want to take a stab at it, might be worth just trying to toss an `#import <locale.h>` at the top of `main.cpp` and `setlocale(LC_ALL "it_IT")` somewhere in `setup`. If this works, I think we could pretty easily expose a setting to configure the locale.

If this doesn't work, I'm afraid the best option is probably to work around it externally. So you'd have whatever is pushing your weather updates also push MQTT variables like `day_0_name`.

tigroz, 2021-05-05:
thank you very much, maybe it is a bit too out of my ability to change the language as you want, since I am here I would like to ask you for another help, I would like to integrate an external temperature sensor but I cannot, this sensor is connected to Home Assistant via zigbee2mqtt, how can I integrate it by replacing the one you entered in the example?
the waveshare display is correctly connected to the mqtt of my server (home assistant) and reads the Openweather weather data but not the data I am trying to get it to read by the temperature sensor
[image]

sidoh, 2021-05-05:
No worries. In the best case, adding support for different locales should be easy. I'll try to find some time to play around with it.

Happy to help with the other question. I'll move it to another issue, though.

sidoh, 2021-05-09:
Unfortunately this didn't work. I'm guessing that the espressif SDK does not ship with other locales.

If anyone knows of a way around this, please let me know.