sidoh/epaper_templates #79: Help displaying temperature from external sensor

sidoh, 2021-05-05:
I would like to integrate an external temperature sensor but I cannot, this sensor is connected to Home Assistant via zigbee2mqtt, how can I integrate it by replacing the one you entered in the example?
the waveshare display is correctly connected to the mqtt of my server (home assistant) and reads the Openweather weather data but not the data I am trying to get it to read by the temperature sensor
[image]

_Originally posted by @tigroz in https://github.com/sidoh/epaper_templates/issues/78#issuecomment-832869192_

sidoh, 2021-05-05:
What do the `json` and `Format, add timestamp` blocks for the `tempetura esterna` look like?

In summary, these should:

1. Extract the value of the temperature
2. Publish a new MQTT message to `epaper-templates/your-display/outside-temp`, with the value being whatever you want the variable in epaper_templates to be (obviously you'll need to substitute the appropriate MQTT topic structure as well)

If you have your sensor set up in HomeAssistant, you could also probably just use a Home Assistant component in Node-RED query the entity state instead of using MQTT. I think the "Poll State" node type would work.

tigroz, 2021-05-06:
thank you very much for your help, I was able to get the external temperature read via zigbee2mqtt ( temp giardino), but I can't get the temperature of the sensor inside the house to read (tado) it always gives me a mistake, obviously since I'm still impractical with NodeRed something wrong ..can you help me?
[image]
[image]

tigroz, 2021-05-07:
these are the flows:
`[{"id":"7dd09f58.f305e","type":"tab","label":"elinkDisplay","disabled":false,"info":""},{"id":"5f0b8ee.a89df7","type":"inject","z":"7dd09f58.f305e","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":145,"y":388,"wires":[["3bd05714.4f59b8"]]},{"id":"998a93fd.f4833","type":"debug","z":"7dd09f58.f305e","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1078,"y":401,"wires":[]},{"id":"76ceb20d.075cfc","type":"mqtt out","z":"7dd09f58.f305e","name":"","topic":"","qos":"","retain":"false","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"b558af9.9ad0b5","x":1091,"y":523,"wires":[]},{"id":"6b34d75b.367358","type":"function","z":"7dd09f58.f305e","name":"Extract Values","func":"var varnamePrefix = msg.varname_prefix || '';\n\nvar messages = [];\nconst addMessage = (varname, value) => { \n    messages.push({\n        topic: `template-displays/bart/${varnamePrefix}${varname}`,\n        payload: value\n    });\n};\n\nconst convertTemp = (c) => {\n   if (c > 200) {\n        c -= 273.15;\n    }\n    return c;\n//    return ((c-32)*5/9);\n//    return ((c*9/5)+32);\n}\n\nconst { \n    temp_max,\n    temp_min,\n    temp\n} = msg.data.main;\nconst icon = msg.data.weather[0].icon;\nconst convertTime = (t) => {\n    var then = new Date(t).getTime();\n    \n    if (then > new Date().getTime()/100) {\n        return then/1000;\n    } else {\n        return then;\n    }\n};\n\n// addMessage('test', msg.data);\naddMessage('forecast_high', convertTemp(temp_max));\naddMessage('forecast_low', convertTemp(temp_min));\naddMessage('forecast_temp', convertTemp(temp));\naddMessage('weather_icon', icon);\naddMessage('timestamp', convertTime(msg.data.dt));\n\nreturn [ messages ];","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":894,"y":480,"wires":[["76ceb20d.075cfc","998a93fd.f4833"]]},{"id":"9fe35092.cc07b","type":"inject","z":"7dd09f58.f305e","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"3600","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":142,"y":317,"wires":[["c83e93b5.032c2"]]},{"id":"986b1975.717578","type":"function","z":"7dd09f58.f305e","name":"Extract Daily Forecasts","func":"const FORECAST_INTERVAL = 3; // hours\nconst NUM_FORECASTS = 5;\n\nconst forecasts = msg.payload;\nconst todaysDate = (new Date()).getDate();\n\nvar messages = [];\n\nconst addMessage = (varname, forecast) => {\n    messages.push({\n        varname_prefix: varname,\n        data: Object.assign({}, forecast, {dt:forecast.dt*1000})\n    });\n};\n\nconst addForecast = (index, forecast) => {\n    addMessage(`daily_weather_slot${index}_`, forecast);\n};\n\n// messages.push({timestamps: {data: forecasts.map(v => (v.dt))}});\n\n// Find first datapoint that's from tomorrow and is after 12pm\nconst firstForecast = forecasts.findIndex(v => {\n    const date = new Date(v.dt * 1000);\n    return date != todaysDate && date.getHours() >= 12 && date.getHours() <= 16;\n});\n\nconsole.log(firstForecast)\n\nvar slotNum = 0;\n\nfor (var i = firstForecast; slotNum < NUM_FORECASTS; i += 24 / FORECAST_INTERVAL) {\n    addForecast(slotNum++, forecasts[i] || forecasts[forecasts.length - 1]);\n}\n\nreturn [ messages ];","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":611,"y":316,"wires":[["998a93fd.f4833","6b34d75b.367358","30de640e.06ebcc"]]},{"id":"6f1354c2.45cdec","type":"debug","z":"7dd09f58.f305e","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":772,"y":636,"wires":[]},{"id":"b9c26cb1.fc39a","type":"function","z":"7dd09f58.f305e","name":"Format, add timestamp","func":"return [\n    [\n        {\n            topic: 'template-displays/bart/outside_temp',\n            payload: msg.payload.temperature\n        },\n        {\n            topic: 'template-displays/bart/outside_temp_updated_at',\n            payload: (new Date()).getTime()/1000\n        }\n    ]\n];","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":590,"y":620,"wires":[["6f1354c2.45cdec","76ceb20d.075cfc"]]},{"id":"3f955b81.b9e9f4","type":"template","z":"7dd09f58.f305e","name":"add sf_ prefix","field":"varname_prefix","fieldType":"msg","format":"handlebars","syntax":"plain","template":"sf_","output":"str","x":364,"y":507,"wires":[["6b34d75b.367358"]]},{"id":"30de640e.06ebcc","type":"debug","z":"7dd09f58.f305e","name":"","active":false,"tosidebar":true,"console":true,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":798.5,"y":238,"wires":[]},{"id":"3bd05714.4f59b8","type":"openweathermap","z":"7dd09f58.f305e","name":"","wtype":"current","lon":"","lat":"","city":"Grottaferrata","country":"Italy","language":"it","x":318,"y":388,"wires":[["6b34d75b.367358","998a93fd.f4833"]]},{"id":"c83e93b5.032c2","type":"openweathermap","z":"7dd09f58.f305e","name":"","wtype":"forecast","lon":"","lat":"","city":"Grottaferrata","country":"Italy","language":"it","x":295,"y":316,"wires":[["986b1975.717578"]]},{"id":"c12c8546.cba2a8","type":"openweathermap in","z":"7dd09f58.f305e","name":"","wtype":"current","lon":"","lat":"","city":"Grottaferrata","country":"Italy","language":"it","x":113,"y":466,"wires":[[ [... cut ...] ,"willQos":"0","willRetain":"false","willPayload":"","willMsg":{},"sessionExpiry":""},{"id":"aa798da0.a36478","type":"zigbee2mqtt-server","name":"","host":"192.168.1.12","mqtt_port":"1883","mqtt_username":"tigroz","mqtt_password":"pepenegro01","tls":"","usetls":false,"base_topic":"zigbee2mqtt"},{"id":"66f8ac9d.0552c4","type":"server","name":"Home Assistant","legacy":true,"addon":true,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":true,"cacheJson":false}]`

sidoh, 2021-05-09:
It looks like you just need to change the code in `Format, add timestamp` to:

```
...
payload: msg.payload
...
```

tigroz, 2021-05-09:
yes! , my bad, sorry :)